iNavFlight/inav-configurator #1428: Change displayed units in metric rendering option

StuweFPV, 2022-01-08:
## Current Behavior
I truly welcome the new rendering option in the configurator. I came across a few "looked over" that i suggest will be adjusted for the next release:

## Steps to Reproduce
1. set "metric" in the rendering option
2. select "Advanced tuning"
3. check "General Navigation Settings"
4. select "Mission Control"
5. create Waypoint
6. activate Waypoint and check "speed"

## Expected behavior
Consistency and realism in the unit conversion

## Suggested solution(s)
Change the following units:
- Advanced Tuning -> General Navigation Settings -> Max. Alt-hold climb rate from **km/h to m/min**
- Advanced Tuning -> General Navigation Settings -> Max. navigation climb rate from **km/h to m/min** 
- Mission Control -> Waypoint configuration -> Speed from **cm/s to km/h**

## Additional context
Climb rates are usually defined in ft/min so i suggest the metric equivalent would be m/min. This is more consistent to the rest and realistic when planning than km/h.
I have not tested the other rendering options. Could be worth a look to see if the same values were overlooked as well.
Overall a great new feature!

---

# Configurator 4.0.0
# INAV/MATEKF765 4.0.0 Dec 15 2021 / 15:15:30 (c20a9852)
# GCC-10.2.1 20201103 (release)

wx4cb, 2022-01-08:
agree with the first two points, as for climb rate it's always in ft or meters per minute... not km/h

for our use thought ft/s or m/sec is acceptable imho

https://skybrary.aero/articles/vertical-speed-indicator

MrD-RC, 2022-01-08:
The vertical speeds should match whatever is used on the OSD for the digital vario. I’ve no idea what that is, as I don’t use metric and am lazy at the moment (need to go out and do shopping, but that’s boring 🤣). If the units in the OSD aren’t suitable, they should be changed too.

StuweFPV, 2022-01-08:
thank you for moving. I wasn't sure where the source is for that. it's no biggie but rather strange so i pointed it out since i know you guys are super busy.

yes km/h is very impractical for use. not even airliners calculate or indicate with such a high unit :) for flight planning m/min would be the most "standard" if you use any aviation tool. but I see the point and agree, that it's probably easier and more in line if it's just the same as in the osd. the osd shows as **m/s**.
thanks!

Mateyhv, 2022-01-14:
In aviation ft/min and m/min are standard, ft/sec and m/sec used currently in Inav are OK either. I can see people benefiting from either method.
I do agree that cm/s is a werid unit as some other, would be greatfor the configurator to have standard units even if the one used by inav are shown in brackets so that the user is aware of the units employed in the background and CLI.

ArnoTlse, 2022-01-14:
Hi,

I will have to have a look of what has been done under the advanced tab to see how we can make things consistent or at least allowing user to configure the unit display. I have added the point to the list.

Regard

Arnaud

MrD-RC, 2022-01-14:
@ArnoTlse if mission control uses the data-setting method to populate the fields, it would be really easy to implement. Just add a `data-unit="cm"` parameter to the input. The unit would be the what is used by the CLI for input. The conversion functions take care of the rest.

ArnoTlse, 2022-01-23:
Hi @MrD-RC,

Just for my knowledge, what is the usage of data-setting compared to data-unit ?
I'm not seeing where data-settings field are used or declared then.

For data-unit, if I have understood well, I just have to update the different fields of Mission planner with the Data-unit field corresponding to the unit INAV expects, and the settings class will display the unit in the system selected by the user in the General Inav settings menu ? 
Then, is there still a need to keep the INAV real unit used in the label of the field ? Is the settings class will display the INAV real unit if "None" unit is selected in the General Inav settings menu ?

Regards

Arno

MrD-RC, 2022-02-01:
@ArnoTlse sorry for the late reply. The _data-setting_ field is how the input is populated, rather than using the MSPHelper. For example on the Advanced Tuning html page, the code for the loiter radius is:
`<input id="loiterRadius" type="number" data-unit="cm" data-setting="nav_fw_loiter_radius" data-setting-multiplier="1" step="1" min="0" max="30000" />`

`data-setting="nav_fw_loiter_radius"` takes care of how the data is populated in the input and saved back to the FC. _nav_fw_loiter_radius_ is the name in settings.yaml, the same as used to access the parameter in the CLI. You can also see the `data-unit="cm"` here which is what is used to take care of the units conversion. _cm_ being the units used in CLI.

For the units conversion to work, the _data-setting_ method must be used to populate the fields. It doesn't work with the MSPHelper method.

ArnoTlse, 2022-02-01:
Hi @MrD-RC,

Ok, thanks. The settings.yaml is the key I've missed. I was looking in Inav Configurator such this mapping definition, it was under INAV firmware side.

Thus, for fields used by INAV firmware, I'll try to update the code with data-setting. But there will be few fields not linked to INAV firmware, but used to set things in the mission planner such custom safehome range radius, default Altitude or speed, that will not been able to use this data-setting stuff I guesss as there are not declared in the INAV firmware settings.yaml file. Maybe it is specific to Mission planner. I will see how to manage this.

closed 2022-02-21 by MrD-RC