iNavFlight/inav-configurator #1431: Make "Manual" as default and "Acro" as mode

StuweFPV, 2022-01-09:
## Current Behavior
Just watched Darren's latest vid and it brought this "old issue" to my mind again. Something i never understood (but maybe i'm missing some info's here) is why is Acro default without any mode assigned and Manual has to be a selected mode? Just by the natural logic it should be the other way around. You do nothing on your FC and you'll get nothing. I remember spending hours searching of the Acro mode at my beginnings, wondering why i had so little control throw when with the servo tester it was all fine. Learning that i was forced to assign a switch just to get the plane flying manual. It has slightly improved now due to the new flag in the configurator but still. Why is Acro being treated so differently from any other mode?

When I setup a new plane i do the mechanics first, then move onto the computer. With inav you first have to get the FC up and running, assign a switch, set modes and switch to manual before you can adjust your servos properly. it's a back and forth just because of that. Also for new guys, if they forget or don't want to start with a mode and just connect the FC and go flying, by default they're in Acro mode. If they forgot to check the direction of the stab correction they can easily loose the plane. I wonder how many problems this design has caused over the years worldwide? And i just can't see why? what is the justification for causing so many unnecessary problems by defying any natural logic?

## Desired Behavior
If you do/select nothing you'll get nothing. You don't even have to assign a switch. If you want something out of your FC like stabilized or autonomous flying then you have to get active, start configuring, learn more, tune properly.

## Suggested Solution
Make "manual" the mode if no mode is selected. make "Acro" a standard normal mode that can be selected like any other mode in the list and remove the newly added flag at the bottom left.

## Who does this impact? Who is this for?
Any users using inav - especially new starters.

wx4cb, 2022-01-09:
i think this would also make the horizon drift issue where something happens and people dont have manual specifically set a help because then if they do have an issue, it's going to affect acro as well is it not? be better to be able to go "oh crap" and disable all flight modes.

If that makes sense

MrD-RC, 2022-01-09:
Because this is a multi-platform firmware, and quads don’t have manual. Acro is the least control to the flight controller that all platforms can use.

closed 2022-01-09 by MrD-RC

StuweFPV, 2022-01-09:
ok well at least i understand now and it makes sense. so it is what it is. another reason to split wings from quads.
thanks for the info!