iNavFlight/inav-configurator #1435: OSD "open font file" messes up regular OSD

StuweFPV, 2022-01-12:
I'm not sure if this is a bug or if I'm doing something wrong here. This is what I did for reproduction:

- downloaded the latest configurator code from github (only found 4.1.0 - no 4.0 available)
- got my desired font file *.mcm
- changed 2 existing icons to my liking with the help of Max4756 editor
- saved the font under a new name but also under the identical name
- connected my plane and went to OSD tab
- opened "font manager" and selected "open font file"
- uploaded the changed font file. 1st test new name, 2nd test original name

It worked well on the screen as my changed icons appeared the way i wanted. also when turning on the plane it all looked good BUT lots of other stuff in the OSD was messed up like no more AHI. On the line of Roll PID the I and D were suddenly missing. etc.

I did not compile a new configurator with that font file. But why would I have to if there is this "open font file" button? I can understand if i would add new icons where the function is missing or start changing the order. but simply changing a existing one... is there a checksum that has to be correct? If so then what is the "open font file" here fore? What am I missing or doing wrong for just change the look of a existing icon? Why are the new icons work but suddenly other stuff starts disappearing especially the AHI?

reapplying the original font again got the OSD back into normal stage.

OptimusTi, 2022-01-13:
You can't modify the font files directly. Read the docs...

StuweFPV, 2022-01-13:
would love to. spent hours looking for anything.. can you point out the one relevant one to me please? Then what is "open font file" here for? who uses that and what for?

StuweFPV, 2022-01-13:
Thank you very much! I still don't get what the "open font file" button is for...

closed 2022-01-13 by StuweFPV