iNavFlight/inav-configurator #1436: Support MSP over UDP

digitalentity, 2022-01-17:
There are multiple solutions to provide connectivity to the FC via WiFi, i.e. https://ardupilot.org/copter/docs/common-esp8266-telemetry.html

Typically they encapsulate telemetry packets (MAVLink) into UDP or TCP and FC doesn't even know that it's talking over the wireless connection.

However, on the Configurator side support for MSP-over-UDP is necessary (as the bare minimum). Corresponding ESP8266 or ESP32 firmware can provide the access-point functionality and UDP endpoint to wrap/unwrap the MSP packets and talk to the FC over the conventional UART.

digitalentity, 2022-01-17:
@MATEKSYS

wx4cb, 2022-01-17:
to be honest, TCP would be the better way of doing this... UDP as a fire n forget protocol could cause more hassles than it's worth.

the data rate isnt that high to be honest and easily handled by a TCP connection

digitalentity, 2022-01-17:
TCP or UDP doesn't matter much IMO. WiFi is acknowledging medium by itself (it's ACKing every frame, except broadcast, on the medium control level), so even in case of UDP that would be pretty reliable. Also, the existing MSP stack in the configurator is retriable and stateless, so this shouldn't be a problem even if some of the packets are lost. But supporting both TCP and UDP would indeed be beneficial and IMO not so different from the implementation perspective.

MATEKSYS, 2022-01-17:
via UDP with ESP8266-wifi-temeletry, very smooth.
https://user-images.githubusercontent.com/23047106/149765209-e2528e17-81ef-4e1d-ba22-db00934ced7e.mp4

wx4cb, 2022-01-17:
@MATEKSYS nice. 

@digitalentity i've used an ESP in there hooked up to a uart in wifi to serial mode for the (similar as to how we hook one up to crossfire wifi for the mini crossbow AAT), and i've found TCP mode to be more reliable even at 5 feet away from the airframe, but obviously supporting both is trivial.

digitalentity, 2022-01-17:
Either way - this is about supporting MSP-over-network in Configurator, which I believe will mostly change the UI and the serial connection layer. We might need custom MSP-aware firmware for the ESP8266 though.

FPVZaphod, 2022-01-19:
If UDP or TCP becomes a bit more common for us iNav folks, an enlightened person might invest time to create an iOS version? Then we don't depend on not existing serial support of iOS anymore. Native configurator on iPad  or iPhone would be amazing!

stronnag, 2022-02-04:
This (used to) work using esp8266 and e.g. `socat` with the configurator 'manual device' in c. 2016. No doubt improvements in `nw.js` have broken this since.
```
 exec socat  pty,link=/tmp/vc0,raw  udp-datagram:esp-air:14014,bind=esp-gcs:14014
```
connect the configurator to `/tmp/vc0` and marvel at MSP over UDP. Similar serial to IP tools no doubt exist for other OS.

Scavanger, 2022-02-11:
I have a first version (via TCP) ready:

https://youtu.be/lnaKceeNjiY

I wrote my own firmware for the ESP8266 ("MSPWifiBridge", see video) that checks the MSP frames for the correct checksum and sends the frames "in one piece", it also provides an easy to use web interface for the settings.
But any other firmware that passes 1:1 data from the Uart works just as well.

The code is based on my BLE support (#1441) but as long as the PR is not yet merged (or rejected) I cannot finish it yet.

jbienz, 2022-02-25:
I'll start posting here since #1441 is now closed. Any news on this Scavenger? I got the Bluetooth classic (SPP) route working with an HC-05 tonight. It wasn't terrible, but it doesn't work with SpeedyBee app.

Scavanger, 2022-02-26:
@jbienz 
Actually, everything is ready (BLE, TCP and UDP), but there is a problem that I cannot solve on the configurator page, here is the discussion: https://github.com/iNavFlight/inav/pull/7783

jbienz, 2022-02-27:
I'll follow along both threads. Thank you for pushing forward on this @Scavanger.

closed 2022-05-31 by DzikuVx