iNavFlight/inav-configurator #1438: [Dev] Chrome Debugger mode not working in 5.0.0 version of Inav Configurator

ArnoTlse, 2022-01-23:
Hi,

I'm having trouble to get "inspect code" by right clicking on the Inav Configurator app with the 5.0.0 version of the configurator. In version 3.0.0 it was working fine.  Now, in 5.0.0 version, there is nothing occuring by right clicking.

What I expect like in 3.0.0 version : 
[image]

Does anybody know how to make it work again or if there is another way to debug Configurator when launched ?

ArnoTlse

DzikuVx, 2022-01-25:
This by design. You have to manually install SDK version of NW.js
`npm install nw@0.50.3-sdk --nwjs_build_type=sdk`

closed 2022-01-25 by DzikuVx

ArnoTlse, 2022-01-25:
Hi Pawel,

Thanks for the answer. 
I have several unsuccessful shot to install this version. I have to remove the node_modules folder and reinstall through the command you indicate, and now it works.

Arno