iNavFlight/inav-configurator #1443: Fails to start in Ubuntu 20.04

voxtelnismo, 2022-02-03:
INAV configurator fails to start in Ubuntu 20.04 and some flavors (tested in Kubuntu and Budgie).

Issue: default Linux installation is missing libatomic.so.1

Solution: sudo apt install libatomic1

rjmelchione, 2023-01-21:
Followed the installation instructions on Ubuntu 20.04.5 LTS (including install of libatomic1). However inav-configurtor still fails to start.  

Any guidance/assistance for how to resolve this would be greatly appreciated.

stronnag, 2023-01-21:
Please explain what you mean by "fails to start", in particular please paste the output from running it from the shell prompt.

rjmelchione, 2023-01-24:
[image]

rjmelchione, 2023-01-24:
I
[image]

rjmelchione, 2023-01-24:
I believe I have followed the install instructions properly. However, when I attempt to launch inav-configurator I get an error that the inav-confiurator is not found.

stronnag, 2023-01-24:
You need to learn some basic Linux. The current working directory is not conventionally on `$PATH`.

rjmelchione, 2023-01-24:
Fair enough. I didn't realize the current directory needed to be added to $PATH. I assumed the current director would be in the execution order and did not need to be explicitly added to the $PATH. This is what happens when you assume. I will fill in my knowledge gaps for Linux. Thank you for the assistance.

stronnag, 2023-01-24:
You can run the configurator from it's directory by explicitly specifying the  path, for example:

```
$ cd ~/rc_models/inav/inav-configurator
$ ./inav-configurator
```

If that works, please close the issue.

rjmelchione, 2023-01-24:
Yes, works perfect. Thank you. 

I do not appear to have permission to close the issue. It appears to have been opened by voxtelnismo.

closed 2023-01-24 by voxtelnismo