iNavFlight/inav-configurator #1453: Inav configurator win64 4.1.0 NW.JS

knobbytrav, 2022-02-15:
Hello...when I try to run this on my Samsung windows 11 laptop, I get a screen with a huge NW.JS in the middle and what looks like a compass in the lower left corner.  It won't load normally.  I did get it installed and running on my work PC using the same downloads, so I am not sure what the deal is
[image]
.

Help!

thanks!

MrD-RC, 2022-02-15:
You need to download it from here https://github.com/iNavFlight/inav-configurator/releases

knobbytrav, 2022-02-15:
I just re-downloaded it and tried again with the same results.

DzikuVx, 2022-02-22:
tested win64 version, works just fine.

knobbytrav, 2022-02-22:
> tested win64 version, works just fine.

On windows 11?

DzikuVx, 2022-02-22:
Yes, on windows 11

knobbytrav, 2022-02-22:
It must be something local on my machine... very frustrating. Any suggestions? I've gotten several clean downloads and done several installations. I am at a loss. Computer is showing there are no updates available either.

DzikuVx, 2022-02-23:
Try this https://github.com/iNavFlight/inav-configurator#inav-configurator-start-minimized-what-should-i-do

knobbytrav, 2022-02-25:
Just tried it....I'm still getting the nws screen.  This is driving me bonkers!

MrD-RC, 2022-11-26:
@knobbytrav was this ever resolved? Is 5.1 working ok?

knobbytrav, 2022-11-26:
No... I can't get any versions to work on this machine.  I just tried a reinstall of 5.1 to no avail...and I also just installed 6.1 to see.  End result is the same....the NW JS screen shot I posted is all I ever get with it.

closed 2023-03-22 by DzikuVx