iNavFlight/inav-configurator #1455: Target MatekF405SE Error: Supplied image is larger then flash available on the chip! Image: 528.9 kiB, limit = 512.0 kiB

anton-rothwell, 2022-02-18:
Seems like this target can't be flashed for Matek F405 WSE FC.

MrD-RC, 2022-02-18:
Are you sure you have the right target. The F405 should have 1024kb of flash. A 512kb limit would be an F411 or F722.

anton-rothwell, 2022-02-18:
> Are you sure you have the right target. The F405 should have 1024kb of flash. A 512kb limit would be an F411 or F722.
> 
> Maybe post up a photos of your flight controller.

[image]
http://www.mateksys.com/?portfolio=f405-wse

MrD-RC, 2022-02-18:
I mean a photo of your physical flight controller. I know that target works fine with it’s intended board.

anton-rothwell, 2022-02-18:
> I mean a photo of your physical flight controller. I know that target works fine with it’s intended board.
The photo is the same as the matek link I provided above, there's only one board like that.

MrD-RC, 2022-02-18:
Actually the F722-WPX looks similar.  I wanted a photo of your actual flight controller as confirmation. 

I know the target you are trying to use works fine with an F405-WSE as I flashed it today. Also the F405-WSE has more than 512kb flash.

Please post a photo so we can try and understand what’s happening here. 

Do you get anything back if you type version in to CLI? If so can you post that up too.

MrD-RC, 2022-02-18:
It should be screen printed in clear text on the flight controller. Matek were also pretty good at screen printing the target on the boards too. Usually on the underside or bottom plate.

anton-rothwell, 2022-02-18:
> It should be screen printed in clear text on the flight controller. Matek were also pretty good at screen printing the target on the boards too. Usually on the underside or bottom plate.

The underside is inaccessible - they don't have anything printed on the top, some FCs are printed on the top, some aren't

MrD-RC, 2022-02-18:
It’s on the underside on the F722-WPX

http://www.mateksys.com/?portfolio=f722-wpx

anton-rothwell, 2022-02-19:
The WPX Firmware flashed OK.  One thing I did notice during setup, is the barometer defaults to : SPL06 when it should be DPS310.  It's not something I specifically remember changing when I last set this FC up with a previous INAV.

MrD-RC, 2022-02-19:
Both barometers have the same hardware ID. So it is impossible of iNav to know which is which. The way to tell for sure is the SPL06 has a metal case, whereas the DSP310 is nylon.

closed 2022-02-19 by MrD-RC