iNavFlight/inav-configurator #1463: Master fails to execute on all (64bit) platforms

stronnag, 2022-02-23:
Running current master on any platform (linux64, osx64, win64) results in a stuck window like:
[image]
Untested on 32bit.

The last working build is from 777aa586e3b6793061ca49adda5c67bd73de1f84

However, `git bisect` shows the culprit as
```
$ git bisect bad
093fc22701aac1d1a4b531e9fd84d5477abe8281 is the first bad commit
commit 093fc22701aac1d1a4b531e9fd84d5477abe8281
Author: Darren Lines <darren@darrenlines.uk>
Date:   Sat Feb 12 17:16:31 2022 +0000

    Enhance and extend units conversion
    
    1. Fixed rounding error.
    2. Added conversions for:
    a) vertical speed
    b) metres
    c) temperature
    3. Added support on Failsafe and OSD pages.
    4. Sorted out alignment issues, due to different implementations.

 _locales/en/messages.json         |  10 +-
 js/settings.js                    | 151 ++++++++++++++++++++------
 main.css                          |  14 ++-
 src/css/tabs/configuration.css    |   4 +-
 src/css/tabs/failsafe.css         |  11 +-
 src/css/tabs/firmware_flasher.css |   1 -
 src/css/tabs/osd.css              |   4 +-
 src/css/tabs/pid_tuning.css       |   1 -
 src/css/tabs/transponder.css      |   1 -
 tabs/advanced_tuning.html         |  12 +--
 tabs/failsafe.html                |  22 ++--
 tabs/failsafe.js                  | 218 +++++++++++++++++++++++---------------
 tabs/osd.html                     |  18 ++--
 13 files changed, 299 insertions(+), 168 deletions(-)
```
Although authored prior to 777aa586, the failing commit was merged immediately after it.

DzikuVx, 2022-02-23:
@stronnag it's runnig correctly on my PC. have you run `npm update` ?

stronnag, 2022-02-23:
I think so, I also tried on two VMs (Ubuntu LTS, Debian Test) where it had never been built before via `npm install`. But let me double check.

DzikuVx, 2022-02-23:
Win11 , win64

[image]

DzikuVx, 2022-02-23:
testing with release version
`node .\node_modules\gulp\bin\gulp.js release --platform=win64`

DzikuVx, 2022-02-23:
OK, it's not working when build with `nw-builder`, fails exactly like you showed @stronnag

DzikuVx, 2022-02-23:
yup, let me dig in what changed in NW.js so it's happening

MrD-RC, 2022-02-23:
I get the same npm start works fine. building a release fails.

DzikuVx, 2022-02-23:
Interesting, rolling back to `^0.50.3-sdk` does not resolve the problem

DzikuVx, 2022-02-23:
OK, I know what's wrong, will fix

PR #1467 (2022-02-23): remove not needed dependency
closes #1463

@MrD-RC
closed 2022-02-23 by MrD-RC