iNavFlight/inav-configurator #1469: Inspect no longer accessible in Configurator

MrD-RC, 2022-02-23:
Right clicking in Configurator when using npm start no longer give a menu of Chrome browser commands, including **Inspect**. This seems to be since updating the nw,js version. The feature was working fine before it updated earlier today.

Anyway to get this back? It's very helpful for development.

DzikuVx, 2022-02-23:
You could have just asked :) 
`npm install nw@0.61.0 --nwjs_build_type=sdk`

closed 2022-02-23 by DzikuVx

MrD-RC, 2022-02-23:
Thought it may have been a bug as it was always on by default before. I’ll update my install script. Thanks Pawel.

breadoven, 2022-08-18:
OK what's the trick to getting this to work ? Tried installing the SDK version (61 and the latest 67.1) but can't get the Inspector/Console to appear using F12 or anything else. Pretty difficult to debug without this.

DzikuVx, 2022-08-19:
@breadoven 

```
rm -rf ./node_modules/nw/*
npm install nw@0.61.0 --nwjs_build_type=sdk
```

breadoven, 2022-08-19:
Well tried the above and Configurator wouldn't run at all afterwards (some problem with openlayers). So deleted the node_modules folder and ran `npm install nw@0.61.0 --nwjs_build_type=sdk` which fixed Configurator running ... but still no Inspector using F12/Cntl Shift J ?. 

However, `npm view nw version` returns `0.67.1` rather than the `0.67.1-sdk` I'd have expected so it doesn't appear to have installed the SDK version for some reason. Not sure what the problem is.

breadoven, 2022-08-19:
OK just tried `npm install nw@0.61.0-sdk` and bizarrely that worked, Inspector now appears. Don't know if this means `--nwjs_build_type=sdk` is broken or something just randomly screwed up.