iNavFlight/inav-configurator #1497: Configurator keeps prompting for new version available

StuweFPV, 2022-03-31:
Whenever I work with an older version but the current one, today <=4.0 I get the prompt "New Configurator Version available". Since the 4.1 Configurator is not "backwards compatible" I always use the corresponding configurator to the version on the FC. Owning close to 20 inav airplanes, not all planes are always on the latest release all the time, so working with older configurators is a common thing for me.

It is really annoying to get that prompt EVERY TIME you open any configurator but 4.1. I've tested it all the way down to 2.6 and he always keeps prompting that a new version is out. Of course the "Receive desktop notification when application updates" checkbox is unchecked on all versions i use. 

This "version checking" is also causing a delay for the com port to show when starting the configurator as he is buys going online. This seems to be permanently embedded somewhere, ignoring the setting in the option?

**Please fix for v5.0 and stop this background checking of new versions.** 
Thanks

wx4cb, 2022-03-31:
well technically it's doing what it's supposed to.... as for the delay to getting the com ports, that's been happening for quite a while now.

StuweFPV, 2022-03-31:
what do you mean "what it's supposed to"?? is this per design and not a bug? then why is there a checkbox "notification for updates"??? do you have any idea how annoying this is? after 20x the prompt per day I did realize that there is a update available but i still will not update... what is the logic behind this??
Please put at least a "do not show this again" checkbox in it like most software have or some other method to switch this off.
Thanks

StuweFPV, 2022-03-31:
I have planes still on 2.6.1, some on 3.0, some on 4.0 and very few on 4.1... I try to fly not constantly update my planes. Do you know how much work it is just to update one plane properly with all the changes going on all the time? It would all not be a problem if the Configurator 4.1 could read and config a 2.6 build. Not all new features in the newer versions are for the better or more stable.. there are valid reasons NOT to upgrade and skip a release or two...

the issue is why is the software telling me every time i open it, that there is a new version? why can i not click this away like in any other software? how many times do i have to be remembered that there is a new version? when i work on a plane it is not unusual to reboot 20x - so within 2h i get 20x the "info" that there is a new release... and no option to confirm "yes I'm aware of that - thank you". but then again as you pointed out there is this option that can be ticked in your screenshot so what is that for??? If i uncheck that box then i expect the soft NOT to look for a update or at least not inform me. So if unchecked and it still reminds me of the new version then what is this option for?? it just does not make sense and in 99% of all software i use i get the "do not show this message again" checkbox. 
and you tell me this is per design and not a bug??? If that is true then i do not understand the concept of the option and the constant reminder of the new version - almost as if it was prohibited under punishment to use a old version on the next day a new one is released...

StuweFPV, 2022-03-31:
thank you for your kind reply - sharing your wisdom was very enlightening

MrD-RC, 2022-03-31:
If behind that checkbox is not suppressing new version notifications when it's unchecked. That is a bug. I can see how that would get annoying.

closed 2022-04-12 by DzikuVx