iNavFlight/inav-configurator #1512: INAV 4.1.0 does not update the take off elevation

tguillaume02, 2022-05-07:
When I setup my take off home on mission controller, the elevation is not updated or keep the last elevation point
Exemple: 
Lat: 2.8156422 Lon: 50.0669514

[image]

breadoven, 2022-06-05:
Does the Home elevation update if you click on the "set Home" button or move the Home point around ?

Also, is this a Home loaded from a Mission file or a new Home point placed on the map ? I assume you have an internet connection when this happens ?

MrD-RC, 2022-11-26:
No response from OP in over 6 months. Seems to work fine in current master. Closing.
[image]

closed 2022-11-26 by MrD-RC