iNavFlight/inav-configurator #1527: 5.0 Calibration Page issues - Upside down resets and can't recalibrate

b14ckyy, 2022-05-29:
Calibration page has some problems with the orientation order. 
As soon as I calibrate board upside down all other calibration directions are reset. I have to do upside down calibration first to complete the process. 
If I wanna recalibrate on an already calibrated FC I also have to start with upside down, to get the other orientations cleared. Any other orientation will just instantly give a popup for completed calibration. 

FC: F765-Wing
INAV: 5.1 RC1
Configurator: 5.1 RC1

How to reproduce: 
- setup new FC
- start calibration with any orientation but upside down
- do upside down mid of the process or at last 
- >> All calibration points are reset
- start with upside down and complete the process
- >> calibration complete

Save and reboot and try a recalibration: 
- Click calibrate with any orientation but upside down
- >> all directions are cleared in the GUI
- confirm the info box and calibrate any orientation but upside down again
- >> calibration will be complete immediately
- start calibration with upside down
- complete all other orientations
- >> re-calibration completes

DzikuVx, 2022-05-29:
This is not new release related. This happens when any step results "unexpected" value.

b14ckyy, 2022-05-30:
What exactly does "unexpected´" value mean?
I just retried with a F405SE and it works no matter what order. With the F765-Wing it is reproducible. If there is a bad value during calibration, why does it complete then when I start with upside down? 
This can be really annoying when I never get the calibration done if I don't know in what direction to start :D

closed 2022-06-07 by DzikuVx