iNavFlight/inav-configurator #1533: INAV-Configurator_linux64_5.0.0 rc1 -crashes/halts and fails to open gui

pitts-mo, 2022-05-29:
I am finding the new INAV-Configurator_linux64_5.0.0 rc1 fails to open gui and appears to crash/halt in the console.

[325288:325288:0529/171007.787533:FATAL:double_fork_and_exec.cc(131)] execv /home/testing/Public/inav/INAV-Configurator-5.0.0-rc1/chrome_crashpad_handler: Permission denied (13)
#0 0x7f7acd4afe03 (/home/testing/Public/inav/INAV-Configurator-5.0.0-rc1/lib/libnw.so+0x5820e02)
#1 0x7f7acd4c34ee (/home/testing/Public/inav/INAV-Configurator-5.0.0-rc1/lib/libnw.so+0x58344ed)
#2 0x7f7acd4c42ea (/home/testing/Public/inav/INAV-Configurator-5.0.0-rc1/lib/libnw.so+0x58352e9)
#3 0x7f7acdfb6ccf (/home/testing/Public/inav/INAV-Configurator-5.0.0-rc1/lib/libnw.so+0x6327cce)
#4 0x7f7acdfacff0 (/home/testing/Public/inav/INAV-Configurator-5.0.0-rc1/lib/libnw.so+0x631dfef)
#5 0x7f7acf853536 (/home/testing/Public/inav/INAV-Configurator-5.0.0-rc1/lib/libnw.so+0x7bc4535)
#6 0x7f7acf84c6df (/home/testing/Public/inav/INAV-Configurator-5.0.0-rc1/lib/libnw.so+0x7bbd6de)
#7 0x7f7ac9cdde85 (/home/testing/Public/inav/INAV-Configurator-5.0.0-rc1/lib/libnw.so+0x204ee84)
#8 0x7f7acd023369 (/home/testing/Public/inav/INAV-Configurator-5.0.0-rc1/lib/libnw.so+0x5394368)
#9 0x7f7acd020e6e (/home/testing/Public/inav/INAV-Configurator-5.0.0-rc1/lib/libnw.so+0x5391e6d)
#10 0x7f7acd021894 (/home/testing/Public/inav/INAV-Configurator-5.0.0-rc1/lib/libnw.so+0x5392893)
#11 0x7f7ac9cdc38b (/home/testing/Public/inav/INAV-Configurator-5.0.0-rc1/lib/libnw.so+0x204d38a)
#12 0x7f7ac7a54d90 (/usr/lib/x86_64-linux-gnu/libc.so.6+0x29d8f)

Received signal 6
#0 0x7f7acd4afe03 (/home/testing/Public/inav/INAV-Configurator-5.0.0-rc1/lib/libnw.so+0x5820e02)
#1 0x7f7acd547941 (/home/testing/Public/inav/INAV-Configurator-5.0.0-rc1/lib/libnw.so+0x58b8940)
#2 0x7f7ac7a6d520 (/usr/lib/x86_64-linux-gnu/libc.so.6+0x4251f)
#3 0x7f7acd546ba5 (/home/testing/Public/inav/INAV-Configurator-5.0.0-rc1/lib/libnw.so+0x58b7ba4)
#4 0x7f7acd4c3977 (/home/testing/Public/inav/INAV-Configurator-5.0.0-rc1/lib/libnw.so+0x5834976)
#5 0x7f7acd4c42ea (/home/testing/Public/inav/INAV-Configurator-5.0.0-rc1/lib/libnw.so+0x58352e9)
#6 0x7f7acdfb6ccf (/home/testing/Public/inav/INAV-Configurator-5.0.0-rc1/lib/libnw.so+0x6327cce)
#7 0x7f7acdfacff0 (/home/testing/Public/inav/INAV-Configurator-5.0.0-rc1/lib/libnw.so+0x631dfef)
#8 0x7f7acf853536 (/home/testing/Public/inav/INAV-Configurator-5.0.0-rc1/lib/libnw.so+0x7bc4535)
#9 0x7f7acf84c6df (/home/testing/Public/inav/INAV-Configurator-5.0.0-rc1/lib/libnw.so+0x7bbd6de)
#10 0x7f7ac9cdde85 (/home/testing/Public/inav/INAV-Configurator-5.0.0-rc1/lib/libnw.so+0x204ee84)
#11 0x7f7acd023369 (/home/testing/Public/inav/INAV-Configurator-5.0.0-rc1/lib/libnw.so+0x5394368)
#12 0x7f7acd020e6e (/home/testing/Public/inav/INAV-Configurator-5.0.0-rc1/lib/libnw.so+0x5391e6d)
#13 0x7f7acd021894 (/home/testing/Public/inav/INAV-Configurator-5.0.0-rc1/lib/libnw.so+0x5392893)
#14 0x7f7ac9cdc38b (/home/testing/Public/inav/INAV-Configurator-5.0.0-rc1/lib/libnw.so+0x204d38a)
#15 0x7f7ac7a54d90 (/usr/lib/x86_64-linux-gnu/libc.so.6+0x29d8f)
  r8: 00007ffc05225a00  r9: 0000000000000000 r10: 0000000000000008 r11: 0000000000000246
 r12: 0000000000000006 r13: 0000000000000016 r14: 00007ffc05226c78 r15: 00007ffc05226c80
  di: 000000000004f6a8  si: 000000000004f6a8  bp: 000000000004f6a8  bx: 00007f7ac56f44c0
  dx: 0000000000000006  ax: 0000000000000000  cx: 00007f7ac7ac1a7c  sp: 00007ffc05225930
  ip: 00007f7ac7ac1a7c efl: 0000000000000246 cgf: 002b000000000033 erf: 0000000000000000
 trp: 0000000000000000 msk: 0000000000000000 cr2: 0000000000000000
[end of stack trace] 

I am using:
Ubuntu 22.04 LTS
Linux testing 5.15.0-30-generic #31-Ubuntu SMP Thu May 5 10:00:34 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

I find I am still able to run INAV-Configurator_linux64_4.1.0 without problem.

pitts-mo, 2022-05-29:
Ha,
I see now that @stronnag and @DzikuVx had already touched this issue with #1529 :-) 

executing this command below in the configurator folder allowed 5.0.0 rc1 to launch :-)
sudo chmod 777 chrome_crashpad_handler

Leaving this ticket open for temporary reference.
-p

ThreeBadmice, 2022-06-13:
Snap! Same issue here...  

Ubuntu 22.04 LTS 

Thanks for the heads up @pitts-mo :+1: 

Solved by making the chrome_crashpad_handler file executable. 

**chmod 700 chrome_crashpad_handler**  - might be a bit gentler ;)

RedFraction, 2022-11-11:
same issue, why u dont fix it? Is it so hard?

MrD-RC, 2022-11-11:
What version are you using. If RC1, update to the stable release and see if that fixes it. RC versions should not be used after the stable version is released.

stronnag, 2022-11-11:
This can be addressed in two ways:
* The user reads the installation instructions and acts accordingly; or
* The release maintainer builds the release archive on Linux (then it becomes a non-issue).

MrD-RC, 2022-11-26:
As this issue is closed. I will move this to a discussion.