iNavFlight/inav-configurator #1534: iNav 5.0.0 Modes not showing in Configurator

billjow, 2022-05-30:


DzikuVx, 2022-05-30:
You have "hide unused modes" selected

closed 2022-05-30 by DzikuVx

billjow, 2022-05-30:
As I explained in the text of the original issue, I was showing the modes that were still visible to me and I explained "if I slide the Hide unused modes back, I get other options, but not the nav ones." This issue should not be closed as it has nothing to do with whether the Hide Unused Modes option is selected or not.

billjow, 2022-05-30:
Hi Pawel,

 

I have commented on the original issue, which I can no longer find. (Did you delete it?).

 

I explained in the original text to the issue that I was showing which modes were showing to me i.e. that all my nav modes had disappeared. I further explained that if I slide the Hide Unused Modes off, other modes became available, but not the relevant nav modes.

 

Please re-read my original explanations – this issue should not be closed as it has nothing to do with the Hide Unused Modes slider. 

 

Kind Regards,

 

Bill

 

Bill Jowitt

Tel: 07790 491385