iNavFlight/inav-configurator #1536: INAV 5 CONFIGURATOR Multi WP Mission Fail to load

abimaputra78, 2022-06-01:
**
****

## Current Behavior

INAV 5.0 CONFIGURATOR failed to load and discrepancy of multi WP Mission. Reading with open street map was also diffcult for drone rural flying

## Steps to Reproduce

1. Setup for mission 1 and 2
2. update all and save to eeprom and save to pc
3. Reload with different mark number and sometimes mission 2 was missing 
4. Reload mission file from pc result same no. 3 above
5. No elevation or contour in the map

## Expected behavior

No discrepancies between Planned Missions and saved missions in the pc and the FC

## Suggested solution(s)

1. Using a Topographic Map was a ++ value since elevation and contour available. 
2. Valid saved multi wp mission during reload

## Additional context

---

- FC Board name and vendor: Holybro KAKUTE H7 
- INAV version string: 5.0 RC1

breadoven, 2022-06-04:
> 3, Reload with different mark number and sometimes mission 2 was missing

What do you mean by different mark number and reload ? Reload from where ?

abimaputra78, 2022-06-05:
From saved abc.mission from PC or load from FC (not eeprom one)

breadoven, 2022-06-05:
Multi missions only work if saved to EEPROM. The Wiki description for Multi Missions might help:

https://github.com/iNavFlight/inav/wiki/Navigation-modes#multi-missions

Also you can't use the `Save Waypoint Mission` stick command if a multi mission is loaded otherwise it will overwrite the multi mission with a single mission.

https://github.com/iNavFlight/inav/blob/master/docs/Controls.md

closed 2022-06-07 by DzikuVx

b14ckyy, 2022-06-11:
@breadoven @DzikuVx 
You should open this again. The description of the opener was really bad but I guess I found what he wanted to report and Jonathan has reproduced as well. 

If a Multi Mission is created and uploaded to FC (not EEPROM yet) and then downloaded, it is still a multi mission with selectable mission Index. But then storing the mission to EEPROM removes the Index markers. So after saving and loading from eeprom all multi missions are loaded as a single mission

Mission 1:
[image]

Mission 2:
[image]

Uploaded all missions: 
[image]

Saved to EEPROM and then loaded from EEPROM: 
[image]

abimaputra78, 2022-06-11:
Yes agree with that. 😉

b14ckyy, 2022-06-11:
@abimaputra78 sorry I have deleted my comment because this is actually an INAV bug. Not configurator. https://github.com/iNavFlight/inav/issues/8112