iNavFlight/inav-configurator #1553: 5.0 RC2 - Waypoint upload/download broken in Mission Control - again

b14ckyy, 2022-06-11:
Might be related to #1366

I have a 53 Waypoint mission planned in MWPTools. Upload and download works there without any problems. 
Uploading the mission from a File to the FC and storing to EEPROM seems to work reliable, just takes very long time (21s). But download is corrupted in most cases. 
When I download the mission from FC now in configurator, sometimes the Mission gets corrupted. Like this: 

[image]

Sometimes more settle like this: 
[image]

Downloading in MWPTools within WSL-g works flawless: 
[image]

breadoven, 2022-09-18:
Is this still an issue @b14ckyy or can it be closed ?

b14ckyy, 2022-09-18:
> Is this still an issue @b14ckyy or can it be closed ?

I think we can close. Seems like the error checks are working. The External USB Port FCs still have issues, shown by sometimes long times until WP are transferred but I got no corrupted missions so far, so I guess that's ok

closed 2022-09-18 by b14ckyy