iNavFlight/inav-configurator #1577: No 'Navigation Modes shown in Modes tab

FranzPVal, 2022-07-09:
Configuring my first iNAV multirotor there is a "tiny" issue with the modes tab: The whole navigation modes section is absent! Fixed-wing setup ok. Instead of the needed nav modes there is a single fixed wing setting (turn assist) between Flight Mode Modifiers and Multi-rotor.
Please fix.

MrD-RC, 2022-07-09:
Turn assist is labelling (will be fixed). It is for multirotors and fixed wing. 

Have you fully set up the GPS?

FranzPVal, 2022-07-11:
I've setup GPS a usual + "use for navigation switch" like I've done on my wings.
In the moment of configuration, GPS hardware is not powered (+ I²C magnetometer + baro isn't working). On wings GPS is also offline if I only plug in the FC.

MrD-RC, 2022-07-14:
The baro not working could be the issue in that case. If it's an on-board baro, are you using the correct target?

The GPS doesn't need to be plugged in. Just set on the ports page and enabled and setup on the GPS page.

FranzPVal, 2022-07-15:
Confirmed.
I see the missing modes now, after I plug in a battery first and then connect the iNAV configurator. My baro can't work w/o battery because on the I²C bus there is a magnetometer (inside a GPS mouse) connected. If not powered the I²C bus became unusable.

I suggest to couple the nav modes section only on the "sensor & buses" setting so that I can configure nav modes even with a "red" baro shown in the upper display.
---
Btw. I have more odd and serious issues using my target FC (Holybro Kakute V2)...