iNavFlight/inav-configurator #1587: Custom build with more than 6 outputs breaks mixer tab

Strat00s, 2022-07-23:
I am now playing with my old Omnibus F4 Pro board and I am trying to get 1 motor and 6 servo outputs. By default, this board has 2 motors and 4 servos. To get my desired configuration, I changed the PPM pin to the motor (as it uses a timer that no other pin uses, thus it should be ok), as I use ELRS via UART1. Here is my current `target.c` config:
<details>
  <summary>target.c</summary>
  
```c++
timerHardware_t timerHardware[] = {
    DEF_TIM(TIM10, CH1, PB8,  TIM_USE_FW_MOTOR, 0, 0), // PPM
    DEF_TIM(TIM4,  CH4, PB9,  TIM_USE_ANY,      0, 0), // S2_IN
    DEF_TIM(TIM8,  CH1, PC6,  TIM_USE_ANY,      0, 0), // S3_IN, UART6_TX
    DEF_TIM(TIM8,  CH2, PC7,  TIM_USE_ANY,      0, 0), // S4_IN, UART6_RX
    DEF_TIM(TIM8,  CH3, PC8,  TIM_USE_ANY,      0, 0), // S5_IN // pad labelled CH5 on OMNIBUSF4PRO
    DEF_TIM(TIM8,  CH4, PC9,  TIM_USE_ANY,      0, 0), // S6_IN // pad labelled CH6 on OMNIBUSF4PRO
    DEF_TIM(TIM3,  CH3, PB0,  TIM_USE_FW_SERVO, 0, 0), // S1_OUT D1_ST7
    DEF_TIM(TIM3,  CH4, PB1,  TIM_USE_FW_SERVO, 0, 0), // S2_OUT D1_ST2
    DEF_TIM(TIM2,  CH4, PA3,  TIM_USE_FW_SERVO, 0, 1), // S3_OUT D1_ST6
    DEF_TIM(TIM2,  CH3, PA2,  TIM_USE_FW_SERVO, 0, 0), // S4_OUT D1_ST1
    DEF_TIM(TIM5,  CH2, PA1,  TIM_USE_FW_SERVO, 0, 0), // S5_OUT
    DEF_TIM(TIM1,  CH1, PA8,  TIM_USE_FW_SERVO, 0, 0), // S6_OUT
    DEF_TIM(TIM4,  CH1, PB6,  TIM_USE_LED,      0, 0), // LED strip for F4 V2 / F4-Pro-0X and later (RCD_CS for F4)
};
```
    
</details>

I also changed the number of PWMs (and motors) in `target.h` to 7.

After flashing the firmware, when I enter the mixer tab, the plane image and servo output table are pretty broken. Instead of populating the image and table with servos and coloring them, it adds a single servo to the end and all the others are just white.
<details>
  <summary>Image warning</summary>
  
[image]
    
</details>

Everything seems to work as it should. All servo outputs work just fine and the motor seems to work too (I tried it with a scope and it does send a 400hz signal).
And the board also sees all servos and motors just fine in CLI `resource` response:
```
# resource
IO:
----------------------
...
A01: SERVO6 OUT
A02: SERVO5 OUT
A03: SERVO4 OUT
...
A08: SERVO7 OUT
...
B00: SERVO2 OUT
B01: SERVO3 OUT
...
B08: MOTOR1 OUT
...
```

I don't know if this is a problem on my side (that I have to edit some more files to fix this) or an inav-configurator bug. Anyone has any idea what might be going on?

MrD-RC, 2022-07-25:
I think this is related to the target. The configurator only displays what the firmware tells it. There are other flight controllers with 12 PWM outputs, so the number is not an issue.

The fact it's trying to put Servo 1 on output S7, which is after the defined maximum of 6, makes me think there's an issue with the definitions.

Just out of interest, maybe try moving the 
```
    DEF_TIM(TIM3,  CH3, PB0,  TIM_USE_FW_SERVO, 0, 0), // S1_OUT D1_ST7
    DEF_TIM(TIM3,  CH4, PB1,  TIM_USE_FW_SERVO, 0, 0), // S2_OUT D1_ST2
    DEF_TIM(TIM2,  CH4, PA3,  TIM_USE_FW_SERVO, 0, 1), // S3_OUT D1_ST6
    DEF_TIM(TIM2,  CH3, PA2,  TIM_USE_FW_SERVO, 0, 0), // S4_OUT D1_ST1
    DEF_TIM(TIM5,  CH2, PA1,  TIM_USE_FW_SERVO, 0, 0), // S5_OUT
    DEF_TIM(TIM1,  CH1, PA8,  TIM_USE_FW_SERVO, 0, 0), // S6_OUT
```
block to after the `TIM_USE_FW_MOTOR` definition, but before `// S2_IN`. It may do nothing. But it's strange that the definition for `//S1_OUT` is the 7th definition in the list.

Strat00s, 2022-07-26:
That did the trick. Moving the outputs together fixed it. Now only the last 2 servos are white (as there are no controls on the plane to which they can be assigned in the mixer)

closed 2022-07-26 by Strat00s