iNavFlight/inav-configurator #1597: WRTH by activating WP

FXvBH, 2022-08-03:
When I activate WP, I see WRTH in the goggles. The mission is then not flown. Been trying to figure out what's going wrong for 2 days, but can't find anything. Does anyone have an idea what's going wrong? Everything looks good, I can just arms, no problems. But he doesn't do a mission anymore. Flight controller is MATEK F405SE.

stronnag, 2022-08-03:
You have not loaded the mission into volatile memory.

FXvBH, 2022-08-03:
If you mean "loaded from EPROM to FC" then I did. Also hung on the laptop, tested whether it came into the FC from the EPROM. If the mission with "load from FC" was neatly put on the map, without taking the power off again test flight, and then again WRTH.

stronnag, 2022-08-03:
Blackbox log then please

FXvBH, 2022-08-04:
Hi stronnag, Sunday we will test it and read the black box. You hear.

FXvBH, 2022-08-08:
We read something about converting man.mode to navwp.mode.
This can cause a problem. Now from man.mode to hor.mode and then navwp.mode.

We didn't get WRTH in the goggles now, but wp. So an improvement

cruise motor up didn't really help.

But the plane wobbled quite a bit. Engine not powerful enough? (Volantex asw28)

Tuesday we do the matek405-se wing in another plane, you hear.

FXvBH, 2022-08-08:
H Product:Blackbox flight data recorder by Nicholas Sherlock H Data version:2 H I interval:32 H Field I name:loopIteration,time,axisRate[0],axisRate[1],axisRate[2],axisP[0],axisP[1],axisP[2],axisI[0],axisI[1],axisI[2],axisF[0],axisF[1],axisF[2],fwAltP,fwAltI,fwAltD,fwAltOut,fwPosP,fwPosI,fwPosD,fwPosOut,rcData[0],rcData[1],rcData[2],rcData[3],rcCommand[0],rcCommand[1],rcCommand[2],rcCommand[3],vbat,amperage,rssi,gyroADC[0],gyroADC[1],gyroADC[2],accSmooth[0],accSmooth[1],accSmooth[2],attitude[0],attitude[1],attitude[2],motor[0],servo[0],servo[1],servo[2],servo[3],servo[4],servo[5],servo[6],servo[7],servo[8],servo[9],servo[10],servo[11],servo[12],servo[13],servo[14],servo[15],navState,navFlags,navEPH,navEPV,navPos[0],navPos[1],navPos[2],navVel[0],navVel[1],navVel[2],navTgtVel[0],navTgtVel[1],navTgtVel[2],navTgtPos[0],navTgtPos[1],navTgtPos[2],navSurf H Field I signed:0,0,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,0,1,0,1,1,1,1,1,1,1,1,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1,0,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1 H Field I predictor:0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,4,9,0,0,0,0,0,0,0,0,0,0,0,4,8,8,8,8,8,8,8,8,8,8,8,8,8,8,8,8,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0 H Field I encoding:1,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1,3,0,1,0,0,0,0,0,0,0,0,0,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0 H Field P predictor:6,2,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,3,1,1,1,1,1,1,1,3,3,3,3,3,3,1,1,1,1 H Field P encoding:9,0,0,0,0,0,0,0,7,7,7,0,0,0,0,0,0,0,0,0,0,0,8,8,8,8,8,8,8,8,6,6,6,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0 H Field H name:GPS_home[0],GPS_home[1] H Field H signed:1,1 H Field H predictor:0,0 H Field H encoding:0,0 H Field G name:time,GPS_fixType,GPS_numSat,GPS_coord[0],GPS_coord[1],GPS_altitude,GPS_speed,GPS_ground_course,GPS_hdop,GPS_eph,GPS_epv,GPS_velned[0],GPS_velned[1],GPS_velned[2] H Field G signed:0,0,0,1,1,1,0,0,0,0,0,1,1,1 H Field G predictor:10,0,0,7,7,0,0,0,0,0,0,0,0,0 H Field G encoding:1,1,1,0,0,0,1,1,1,1,1,0,0,0 H Field S name:flightModeFlags,stateFlags,failsafePhase,rxSignalReceived,rxFlightChannelsValid,hwHealthStatus,powerSupplyImpedance,sagCompensatedVBat,wind[0],wind[1],wind[2],IMUTemperature,baroTemperature,sens0Temp,sens1Temp,sens2Temp,sens3Temp,sens4Temp,sens5Temp,sens6Temp,sens7Temp,escRPM,escTemperature H Field S signed:0,0,0,0,0,0,0,0,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1 H Field S predictor:0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1 H Field S encoding:1,1,7,7,7,1,1,1,0,0,0,0,0,0,0,0,0,0,0,0,0,1,0 H Firmware type:Cleanflight H Firmware revision:INAV 5.0.0 (d2adf35c) MATEKF405SE H Firmware date:Jun 20 2022 11:00:33 H Log start datetime:2022-08-07T10:58:59.906+00:00 H Craft name: H P interval:1/32 H minthrottle:1042 H maxthrottle:1850 H gyro_scale:0x3f800000 H motorOutput:1042,1850 H acc_1G:2048 H vbat_scale:110 H vbatcellvoltage:33,35,42 H vbatref:1243 H currentMeter:0,317 H looptime:1000 H rc_rate:100 H rc_expo:70 H rc_yaw_expo:20 H thr_mid:50 H thr_expo:0 H tpa_rate:0 H tpa_breakpoint:1500 H rates:20,20,20 H rollPID:5,7,0,50 H pitchPID:5,7,0,50 H yawPID:6,10,0,60 H altPID:40,5,10 H posPID:75,5,8 H posrPID:0,0,0 H levelPID:20,5,75 H magPID:60 H velPID:0,0,0 H yaw_lpf_hz:0 H dterm_lpf_hz:110 H dterm_lpf_type:2 H dterm_lpf2_hz:0 H dterm_lpf2_type:0 H deadband:5 H yaw_deadband:5 H gyro_lpf:0 H gyro_lpf_hz:25 H gyro_lpf_type:1 H dynamicGyroNotchQ:250 H dynamicGyroNotchMinHz:30 H acc_lpf_hz:15 H acc_hardware:3 H baro_hardware:0 H mag_hardware:0 H serialrx_provider:9 H motor_pwm_protocol:0 H motor_pwm_rate:400 H debug_mode:0 H features:810028246 H waypoints:8,1 H acc_notch_hz:0 H acc_notch_cutoff:1 H pidSumLimit:500 H pidSumLimitYaw:350 H axisAccelerationLimitYaw:10000 H axisAccelerationLimitRollPitch:0 H rpm_gyro_filter_enabled:0 H rpm_gyro_harmonics:1 H rpm_gyro_min_hz:100 H rpm_gyro_q:500 vA!$ vC#$ vC)$ vC3& vC7( vCA* vCE* vAO, vAU. vAY0 vA_6 vAa8 vAe> vAiB vAkD vAoF vAqF vAsH vAuH vCuH vCwH vCwH vE{F gx4I #0ti 5Vti )]ti -,tl [R%I  1to %Ff7 ki(" ik.6 im2< go8> eq<: cq>6 asD. auF* _yN  c{l" a}p& 1o+I V%'I z!!) |))) |33+ 4* D ~;;+ 8,$D :,(D <.(F @.(F B.(F  D.(F "F.(F $H,*F &J,,F *L,.D ,P*0D .R(4D 2T(6D 4V&8D 6Z$:D 8\$<D <^"<D >b <F  4=1 ".;1 $*;1 &&91 (E!-6 *C#-8 ,C%-< ,A'-> .A)-@ 0A--D 0?/-F 2=1-J 4;3-L 677-N 859-R :5;-T <3=-X <1?-\ >/C-^ @-E-d B+G-f D'I-j F%K-n H#M-p J!Q-t L!S-v N!U-z P!U-~ R#W- T#Y- V%Y- X%[- X'[- Z']- \)]- ^)]- ^)]- `+]- ~`+]- |b-]- |b-[- zd/[- xd-Y- vf+Y- vf)W- th'W- rh%U- rj!S- /`.!l /X.#b /V.%^ /P0%X /L0%V /F0!T zz(1 xr(1 tf(1!4% rZ*1%4# nN*1+2! l@*1-2! j0(112 f"(/52 */72 */92 */;2 ,/=2 */A0 */A0 (/C0 $/E0 xqqE rmmG !/K, lkkK %-I, heeQ '+G, faaU )+E, d[[[ +)A, `WW_ +'?* ^OOc #-%=* ^IIg '/#9* \AAk -/#7* \77o 5/!3* \//q \))u \##w d))y f//y l11y r++{ v''} 3~E# =jE# !1:= #1<= '1@; ~)1D; |-1H; z11L; x51N; v91R; r;1T; n?1Z9 lC1^9 hE1`9 fG1d9 fG1f9 fI1l7 dK1p7 dM1r7 `M1x7 \O1z7 XO1~5 $%$" &%$& ,%&, 0#&0 6#&6 <#&< >#&@ D#&H H#(L N#(R T!&X X!&Z ^!$^! `!$`! f!$h! l!$n c &` _"(^ [$(X W&(V S&(P M((L K((J G**F E**D A,*@ =.*< 7.*: 1.*6 +.*4 #.*0 g;&} W;*i O;(c ?;,Q 7;,G %;,7 ,,;( .4;( 4B;"( 6H;$2  Xr; "`x9 $dx9 &l|9 &p|9 n:/xf+ j:1rd+ b:3h`+ \:3b^) T<7XX) J<7NT) F<7HR) <<9>L) 6<98J) ,<;.F) "<;">) +<7%%)%u /<5)-)'} 9<53;)% =<57C)' E<3?S)' O<3Gc)' S<3Ki)! [:3Sw)! _:3Y}) i:1a q:1i u:3o }83y ]K2w IK4_ ?K2Y -K0I 'K&M 8=-1 2=)5 "=%A !w {45 y w45 dc  &G ha &&G la"*&G pa"0(G va"8(G x_":*G ~_">,G _ B.G } [... cut ...] L::/ 5B2)m L::/ 3B2'm J::/ 3@2%m H88/ 1>2#m F66/ 1<2!m D66/ B66/ B44/ @22/ >22/ <00/ :../ 8../ 6,,- 4**- 4((/ 2&&/ 0$$/ 0$$/ .""/ .""/ ,""/ ,""/ ,""/ ,""/ ,""/ ,$$/ &2 m .$$/ &2 m .$$/ &2$m .$$1 (2&m .  1 (2(m (0,m (0,m (00m (00m *04m *06k ,06k ,08k .0:k 00<k 20>k 20@k 20Bk 40Bk 40Di 40Fi 60Hi 60Ji 60Ji 60Li 60Pg 5D$X 76cs 72y1 9,+O 5$6a ,90E 63.$ 4/,I F/,Y H-(s 2+,i> #E0{ -/Je /1P; .3bWk $) b cJJ( uRR, {RR, }PP. {<<, q::, q66& q44$ y@@  %%1% !'5g 1'59 9!7T EDxQ mzzP End of log (disarm reason:4)

stronnag, 2022-08-08:
1. You cannot enter WP mode from manual
2.  Only post BBL as an attachment.

FXvBH, 2022-08-08:
Hi Stronnag.
Thank you.
Greetz Frans

closed 2022-08-08 by FXvBH