iNavFlight/inav-configurator #1616: Access to FC through ELRS RX Wifi

ArnoTlse, 2022-09-11:
Hi,

Following the new [feature of ELRS 3.0 which allows now to use Betaflight configurator connected to the FC wirelessly through ELRS RX AP or Home network](https://github.com/ExpressLRS/ExpressLRS/pull/1274), I have tried to test it also on Inav Configurator.

I have my drone flashed with the latest Inav (5.1) version and my RX/TX also with the latest ELRS 3.0 version. In the INAV configurator, I have selected the TCP protocol and put the IP address of the RX on my home network. I have found a port number in a thread on ELRS. When I do this, It seems to launch properly the TCP connection, but it seems there is no feedback from FC (after 10s).

[image]

I wonder if anyone as succeed to connect INAV configurator to the FC using this features ? Or is there any modification to provide on INAV configurator to make it work in the future ?

Arno

ArnoTlse, 2022-10-29:
Hi,

As someone succeed since to do it?
Or is there some modification to provide to inav configurator code to make it work?

Arno

padcom, 2023-01-22:
Any progress on that one?

sixtyfive, 2023-01-23:
#8691 mentions it is a duplicate of this issue here. But also #8691 is marked as completed. Confused...

MrD-RC, 2023-01-23:
> #8691 mentions it is a duplicate of this issue here. But also #8691 is marked as completed. Confused...

It is closed because it is a duplicate. It is not complete, just closed.

padcom, 2023-01-26:
Cute, https://github.com/iNavFlight/inav/issues/8691 says "Duplicate of https://github.com/iNavFlight/inav-configurator/issues/1616" and this one links back to https://github.com/iNavFlight/inav/issues/8691... or am I missing something?

MrD-RC, 2023-01-26:
> Cute, https://github.com/iNavFlight/inav/issues/8691 says "Duplicate of https://github.com/iNavFlight/inav-configurator/issues/1616" and this one links back to https://github.com/iNavFlight/inav/issues/8691... or am I missing something?

You're misreading. 8691 is a duplicate of this issue. Buy mentioning that it is a duplicate of this issue, it creates a reference from this issue to the duplicate. It is not looping back. They duplicate issue has been closed.

x8n1620, 2023-04-18:
Is there any progress on this Topic?
I like the way to configure my betaflight Quads via WiFi and would be lucky to have it on my inav-wings too.

Belrmar, 2023-12-31:
On sealed builds like boats this is a big issue for me, every time i want to configure or make some mission planning i have to remove a gasket and several screws, wifi would solve a big issue for those kinds of builds

closed 2025-01-18 by b14ckyy