iNavFlight/inav-configurator #1618: OSD Elements Causing Settings Issue

joshfisk3, 2022-09-14:
Flight Controller: Matek F722 WPX
Inav Version: 5.0.1
Configurator Version: 5.0.1
OS: Windows 11

Changing element items while having the HD video format option selected and using HDZero VTX on a port seems to have issues first saving and secondly will occasionally reset many of the settings on the flight controller as well as causing the settings validated pre-arm check to fail. This issue seems to happen mostly when changing the system messages and fly/on-time timer osd elements. Any change to these elements causes the issue described.

The receiver channel map is reset to RET and Receiver Mode section is completely cleared out.
[image]

[image]

MrD-RC, 2022-09-15:
I don't have HDZero. But have enabled HDZero in ports, and switched to HD on the OSD Page. I can't replicate this problem at all. I have disabled elements, moved them, had them straddling the red line. Nothing wipes any settings, nor locks up the CLI.

All I can recommend trying is:
- Make sure you are using the correct target. For that FC it's pretty obvious (**MATEKF722WPX**) so I don't think its' that. But always worth checking.
- Make sure you have switched off the FrSky Pixel OSD on the ports page. Maybe having two external OSDs on UARTs could be causing strange things to happen.

joshfisk3, 2022-09-15:
I think you were right on the Frsky Pixel OSD. After removing that seems to be working just fine as well as helped with wtfos issues I was having. Thank you for that advice.

closed 2022-09-15 by joshfisk3