iNavFlight/inav-configurator #1622: [VIDEO] [POC] Flight controller configuration is erased when throttle is 100%

Mystraht, 2022-09-18:
Hello,

When you test your receiver, in receiver tab, if you put throttle to 100% (stick up) and you play a little bit with right stick (pitch roll) your configurator interface will freeze/lag, if you disconnect or remove your usb cable, it will wipe all your flight controller data.

How to reproduce :

1. Go to receiver tab
2. Turn on your radio
3. Put the throttle to 100%
4. Put roll to 100% (to the right)
5. Wait few seconds
6. Unplug usb cable from the flight controller
7. Connect again
8. Your flight controller configuration is wiped

Video : https://www.youtube.com/watch?v=X5oZdWBok90

wx4cb, 2022-09-18:
wouldnt that be because you're doing the stick commands and then unplugging while it's probably trying to save something ?

Mystraht, 2022-09-18:
No I did many test, when you do that it seem the program enter in an infinite loop (all the interface is very laggy or frozen in extreme case) when it's in that state, you are already screwed because even if you release all the stick, the FC will be wiped once you disconnect the cable

You can see the MSP Roundtrip counter go very high when you do the procedure I just wrote, which is not working as intented

wx4cb, 2022-09-18:
what happens if you just leave it for a while? does it "calm back down"? if could be trying to format the SD card or flash or something like that?

MrD-RC, 2022-09-18:
That would be trimming the accelerometer, if the stick commands work while connected to USB. That's something that can be checked.

OptimusTi, 2022-09-18:
Is there a way to disable stick commands from the CLI?

Mystraht, 2022-09-18:
> what happens if you just leave it for a while? does it "calm back down"? if could be trying to format the SD card or flash or something like that?

Yep it calm back down unless to stay too much time in that "buggy" state, at a point you reach a no-return situation

MrD-RC, 2022-09-18:
> Is there a way to disable stick commands from the CLI?
> 
> 

Yes, I did that for the profile switching Logic Conditions. They're disabled when connected to Configurator.

MrD-RC, 2022-09-18:
So, stick commands are active. I've have a[ branch where they will be disabled when connected to CLI](https://github.com/iNavFlight/inav/tree/MrD_disable-stick-commands-if-connected-to-configurator). But more may need to be done to have them disabled when connected to Configurator as a whole. I'll get it tested soon.

Mystraht, 2022-09-18:
I don't know if it's a good thing to disable them when your are connected to configurator, it can be useful (like testing elevons on a wing)
I think the best thing here is to find out why there is a bandwidth overload when you put throttle to 100 and move other sticks around 🤷

wx4cb, 2022-09-18:
> I don't know if it's a good thing to disable them when your are connected to configurator, it can be useful (like testing elevons on a wing) I think the best thing here is to find out why there is a bandwidth overload when you put throttle to 100 and move other sticks around 🤷

no that should still. work, he's talking about the stick commands like for changing trims, saving etc. there's no need for them to be enabled in the fc if connected to the configurator

Mystraht, 2022-09-18:
> > I don't know if it's a good thing to disable them when your are connected to configurator, it can be useful (like testing elevons on a wing) I think the best thing here is to find out why there is a bandwidth overload when you put throttle to 100 and move other sticks around 🤷
> 
> no that should still. work, he's talking about the stick commands like for changing trims, saving etc. there's no need for them to be enabled in the fc if connected to the configurator.
> 
> as far as the overload, it's because the FC it trying to do what you requested and set trims etc, stopping stick commands in the configurator will eliminate that

Oh I see thanks, I did not understand that you were talking about that from the beginning!

MrD-RC, 2022-09-18:
> 
> Oh I see thanks, I did not understand that you were talking about that from the beginning! 

Yes, that's what we've been talking about from the beginning. When you hold the sticks in a certain position, they will activate commands to the flight controller. https://github.com/iNavFlight/inav/blob/master/docs/Controls.md#stick-positions

Moving the sticks to see the ranges on the receiver page and move the control surfaces definitely still needs to work.

MrD-RC, 2022-09-19:
Unfortunately, that does only work in the CLI, not configurator as a whole. 

I've submitted a PR, as it will at least disable them in the CLI. I also added an OSD message when settings get saved. I used it for testing. But figured it would be useful anyway.