iNavFlight/inav-configurator #1646: Loss of Navigation modes in modes-tab configuration when a change is made with usb-connected multirotor not powered

alccrl, 2022-11-03:
Configuration :
Inav (configurator) 5.1 or 6.0 .
Target : multirotor with Matek F722STD FC
RC : Taranis x9 Lite with OpenTx.
Laptop : Linux or Win 10 (same behaviour)

What happens: 
existing navigation modes configuration is lost (e.g. POSHOLD/RTH setup) if a change in modes tab ranges is made with a connected usb quad not powered.
 
Comment:
In spite of the requirement sentence (nav modes wiki explanation):
(https://github.com/iNavFlight/inav/wiki/Navigation-modes)
---------------
"For safety reasons, INAV’s navigation modes can be activated only if:

ACC and MAG (multirotor only) are calibrated properly
a valid 3D GPS fix is available
a valid altitude source is available
the FC is armed
This applies to enabling the navigation modes in the Configurator as well as at the flying field. (For bench tests without(!) propellers you may change “set nav_extra_arming_safety = ON” to “OFF” in CLI.)" ....(...)
---------------------

loss of configuration should not happen if a NAV configuration IS >>ALREADY EXISTING<<.

I had a serious quad crash because I started RTH from RC after video loss, but the quad did not start RTH, all opentx logs and blackbox logs demonstrates that NAV modes where absent.

Alfredo

Jetrell, 2022-11-11:
I haven't had this issue with any of the newer Matek boards. That have up to 3 UART ports, that can supply 4.5volts to the GPS/Mag and even drive 3.3v for the Baro, from the USB power.

But on any older Matek stuff.. OR many other flight controllers from other manufactures. That were predominately designed for Betaflight. Then a target was written for INAV....  Can be a crash waiting to happen. Unless you check your setting after every change.

I haven't had them disappear, and not reappear when the battery was plugged in to power the Baro or GPS.
But what has happen to me. Is I saved some other changes I made and overwrote the old Diff All file.. Which now have all the GPS modes missing.. 
So next time I load that Diff All. I had no RTH, Poshold or WP modes, like you.

I recognize this is partially a hardware fault. 
@alccrl But I agree that it would be better if the `navigation modes` stayed put, even when a sensor isn't detected on connection to the Configurator.

simoccc, 2023-04-21:
Hi, same here , I use TMOTOR F7 PRO full function FC and it got no powered pad sadly.

I think that this is quite dangerous, especially on the filed with speedy bee "like" app and connections to FC without battery can lead to inadvertently change settings and fly afterwards without RTH working. That is what happen to me today , luckily I always check as soon as I'm starting the flight .

I guess this issue is on all FC that does not have powered 5v FC pads.

If it's not too complicated that would be a very nice fix

DzikuVx, 2024-04-09:
solved by https://github.com/iNavFlight/inav/pull/9913

closed 2024-04-09 by DzikuVx