iNavFlight/inav-configurator #1651: iNav 6.0.0 - set nav_land_minalt_vspd = 49 not saving/defaulting back to 49

cpwilson, 2022-11-20:
## Current Behavior
Configured iNav 6.0.0 to try and on arming get message that "Value nav_land_minalt_vspd out of range

In CLI value for 'set nav_land_minalt_vspd = 49

Changed to 50 and saved in CLI and able to arm as expected.     Arrived at flying site and same message again on arming, "Value nav_land_minalt_vspd out of range
Checked via OSD Menu and it's defaulted back to 49, changed to 50 again via OSD Menu and saved, able to arm.

Flew and autotune, well impressed and horizon drift gone :) awesome work Devs !!!!

Came home and again in CLI value for 'set nav_land_minalt_vspd = 49' and unable to arm.

## Steps to Reproduce
Power cycle of FC seems to default 'set nav_land_minalt_vspd' to 49

- FC Board name and vendor:
Matek F405WSE

- INAV version string:
# version
# INAV/MATEKF405SE 6.0.0 Oct 20 2022 / 15:20:51 (d1f6f19a)
# GCC-10.2.1 20201103 (release)

Status Output showing error:
# status
System Uptime: 166 seconds
Current Time: 2022-11-20T13:14:14.654+00:00
Voltage: 0.14V (1S battery - NOT PRESENT)
CPU Clock=168MHz, GYRO=MPU6000, ACC=MPU6000, BARO=DPS310
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK, IMU2=NONE
SD card: Manufacturer 0x2, 3858432kB, 10/2010, v0.6, 'SA04G'
Filesystem: Ready
Stack size: 6144, Stack address: 0x10010000, Heap available: 1628
I2C Errors: 0, config size: 9010, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = none, used = none
System load: 7, cycle time: 1005, PID rate: 995, RX rate: 9, System rate: 9
Arming disabled flags: NAV RX CLI SETTINGFAIL
### ERROR: Invalid setting: nav_land_minalt_vspd

MrD-RC, 2022-11-20:
Had you gone in to the advanced tuning page between the FC working and the setting being wrong?

Im also assuming that Configurator is using Imperial or OSD (set to Imperial, UK, or GA) units.

cpwilson, 2022-11-20:
Hi Darren
Yes, If I go into Advanced Tuning page, change nothing, then do a save and reboot it defaults back to 49
If I don't open the  Advanced Tuning page and just do a cli save it stays at 50, what I change it to.

Configurator option is set to imperial and OSD set to UK

MrD-RC, 2022-11-20:
Perfect. That means the PR will fix the issue. 

I was a little concerned when it sounded like you just powered off and on, then the setting was changed.

jayss66, 2022-11-27:
I have the exact problem, I thought I had fixed that problem last week, until I went to fly today and the same error message showed back up. I thought I was losing my mind. lol

PR #1649 (2022-12-04): Improve non-metric handling in units conversions
Fixes #1651 

- Forced default min/max after conversion back to firmware units
- Increased decimal places for non-integer conversions
- Fixed a typo
closed 2022-12-04 by MrD-RC