iNavFlight/inav-configurator #1663: Calibrate Accelerometer button not shown for a newly flashed board

xpol, 2022-12-07:
I have tested with `MATEKH743`, `JHEH7AIO` and my own porting in progress board iFlight Whoop F411 boards, the newly flashed boards will have `Run-time calibration` check in red.
With the great help of *b14ckyy* on Discord, we found that the `Reset Accelerometer Calibration` is shown instead of `Calibrate Accelerometer` button.

Later I found that the 6.0 FP-1 version of inav-configurator is working and shows the `Calibrate Accelerometer` button correctly.

After a few steps of `git bisect`, I found that the commit 1767ed31367f002d2e73ecfc12cb72fcd759bbbe introduce the issue.

Unfortunately, I have no knowledge to understanding the code:
```javascript
        let showCalibrate = false;
        var flagNames = FC.getArmingFlags();
        if (flagNames.hasOwnProperty(13)) {
            if (bit_check(CONFIG.armingFlags, 13)) {
                showCalibrate = true;
            }
        }
```

So I raised the ticket here, I hope it would help you guys to improve the logic here for the not-yet calibrated boards.

MrD-RC, 2022-12-07:
@xpol Could you do me a favour and post the cli you get back after entering these two commands:
```
get accgain
get acczero
```

closed 2022-12-08 by MrD-RC

xpol, 2022-12-09:
@MrD-RC 

```bash
# get accgain
accgain_x = 4096
Allowed range: 1 - 8192

accgain_y = 4096
Allowed range: 1 - 8192

accgain_z = 4096
Allowed range: 1 - 8192

# get acczero
acczero_x = 0
Allowed range: -32768 - 32767

acczero_y = 0
Allowed range: -32768 - 32767

acczero_z = 0
Allowed range: -32768 - 32767
```

MrD-RC, 2022-12-09:
@xpol Thanks. I've no idea why the original check didn't work in that case. That's definitely not calibrated, so the arming check should have failed. Anyway, the new update will fix that. So all good in master now.

PR #1664 (2022-12-08): Fix calibration button
Fixes #1663 

There was an issue with the calibration button on _some_ flight controllers; after performing a fresh installation of INAV. This issue did not appear on my test F405-WING. I have moved the check to the FC object, and added extra checks for the default accelerometer values. Works as expected on my test FC.