iNavFlight/inav-configurator #1682: Cannot run inav configurator 5.1 on Ubuntu 22.10

SergeiDelov, 2022-12-28:
inav terminal window log:
serge@ubuntu-amd:~$ cd Projects/INAV-Configurator5.1/
serge@ubuntu-amd:~/Projects/INAV-Configurator5.1$ ./inav-configurator 
[6957:6966:1227/235600.744898:ERROR:cert_verify_proc_builtin.cc(603)] No net_fetcher for performing AIA chasing.
[6957:6966:1227/235600.775405:ERROR:cert_verify_proc_builtin.cc(603)] No net_fetcher for performing AIA chasing.
[6924:6938:1227/235603.030469:ERROR:chrome_browser_main_extra_parts_metrics.cc(227)] START: ReportBluetoothAvailability(). If you don't see the END: message, this is crbug.com/1216328.
[6924:6938:1227/235603.030484:ERROR:chrome_browser_main_extra_parts_metrics.cc(230)] END: ReportBluetoothAvailability()
[6957:6966:1227/235605.008042:ERROR:cert_verify_proc_builtin.cc(603)] No net_fetcher for performing AIA chasing.
[6957:6966:1227/235703.087471:ERROR:cert_verify_proc_builtin.cc(603)] No net_fetcher for performing AIA chasing.

Log from inav window:
2022-12-27 @ 23:56:01 -- Loaded release information from GitHub.
2022-12-27 @ 23:56:07 -- Connected to: /dev/ttyACM0
2022-12-27 @ 23:56:07 -- MSP connection successfully opened with ID: 1
2022-12-27 @ 23:56:17 -- No configuration received within 10 seconds, communication failed
2022-12-27 @ 23:56:17 -- MSP connection successfully closed

stronnag, 2022-12-28:
Works fine here ....

* You've followed all the requirements of the README And Release Note?
* You're a member of the `dialout` group?

[image]

SergeiDelov, 2022-12-28:
Hi Jonathan.
I added myself to the `dialout` group.
Just for a test, I just run `inav-configurator` as a root, with the the same result.

From the info I provided it is obvious configurator can connect to the port:

`2022-12-27 @ 23:56:07 -- MSP connection successfully opened with ID: 1`

However, there is a problem with SSL certificate:
`6957:6966:1227/235605.008042:ERROR:cert_verify_proc_builtin.cc(603)] No net_fetcher for performing AIA chasing.`

[image]

Thank you.

stronnag, 2022-12-28:
I doubt the SSL error is relevent.

Can you post the connection info from `dmesg`,  which should look something like:
```
$ dmesg
...
[  565.125800] usb 1-1.4: new full-speed USB device number 4 using uhci_hcd
[  565.316149] usb 1-1.4: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 2.00
[  565.316166] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  565.316187] usb 1-1.4: Product: FURIOUS F35-LIGHTNING
[  565.316192] usb 1-1.4: Manufacturer: INAV
[  565.316197] usb 1-1.4: SerialNumber: 0x8000000
[  568.666238] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
[  568.666409] usbcore: registered new interface driver cdc_acm
[  568.666419] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[ 1102.758830] usb 1-1.4: USB disconnect, device number 4
...
```
and also confirm the and group:
```
$ id
uid=1000(jrh) gid=1000(jrh) groups=1000(jrh),20(dialout)
$ ls -l /dev/ttyACM0
crw-rw---- 1 root dialout 166, 0 Dec 28 07:51 /dev/ttyACM0
```
Can you enter the CLI from a terminal program (e.g one of `minicom`, `picocom`, `cu`, mwptools' `cliterm`) ?

stronnag, 2022-12-28:
You may also find the following useful (or not ...)

https://github.com/stronnag/mwptools/wiki/Serial-USB-RAQ

https://github.com/stronnag/mwptools/wiki/Serial-USB-checklist

SergeiDelov, 2022-12-28:
Jonathan,
It is **not a serial USB port issue**. I just discovered exactly the same problem on Ubuntu 22.04.

It is configurator 5.01 issue.

Older versions (3.02, 4.0) of inav configurator work, both connect to FC:

    # version
    # Betaflight / MATEKF405 (MKF4) 3.2.0 Oct 12 2017 / 15:59:13 (f2f34b4) MSP API: 1.36

SergeiDelov, 2022-12-28:
`28/12/2022 11:25	kernel	usb 1-2: new full-speed USB device number 3 using xhci_hcd`
`28/12/2022 11:25	kernel	usb 1-2: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 2.00`
`28/12/2022 11:25	kernel	usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3`
`28/12/2022 11:25	kernel	usb 1-2: Product: MatekF4`
`28/12/2022 11:25	kernel	usb 1-2: Manufacturer: Betaflight`
`28/12/2022 11:25	kernel	usb 1-2: SerialNumber: 0x8000000`
`28/12/2022 11:25	mtp-probe	checking bus 1, device 3: "/sys/devices/pci0000:00/0000:00:08.1/0000:04:00.3/usb1/1-2"`
`28/12/2022 11:25	mtp-probe	bus: 1, device: 3 was not an MTP device`
`28/12/2022 11:25	kernel	cdc_acm 1-2:1.0: ttyACM0: USB ACM device`
`28/12/2022 11:25	kernel	usbcore: registered new interface driver cdc_acm`
`28/12/2022 11:25	kernel	cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters`
`28/12/2022 11:25	mtp-probe	checking bus 1, device 3: "/sys/devices/pci0000:00/0000:00:08.1/0000:04:00.3/usb1/1-2"`
`28/12/2022 11:25	mtp-probe	bus: 1, device: 3 was not an MTP device`

SergeiDelov, 2022-12-28:
`uid=1000(serge) gid=1000(serge) groups=1000(serge),4(adm),20(dialout),24(cdrom),27(sudo),30(dip),46(plugdev),122(lpadmin),132(lxd),133(sambashare)`

`crw-rw---- 1 root dialout 166, 0 Dec 28 12:00 /dev/ttyACM0`

stronnag, 2022-12-28:
> `28/12/2022 11:25	kernel	usb 1-2: Manufacturer: Betaflight`

Maybe you should flash INAV .... perhaps using `dfu-util`

SergeiDelov, 2022-12-28:
Flashed 5.1 firmware using inav-configurator 5.1 GUI. Now inav-configurator connects to FC.

Summary:  If the board has some (old) Betaflight firmware, inav configurator refuses to connect with some SSL certificate error.

Thank you for your time, Jonathan.

closed 2022-12-28 by SergeiDelov