iNavFlight/inav-configurator #1687: can not communicate with the FC

csessa5, 2023-01-10:
I have downloaded Inav for iMac.  it all seems to work up to the point of flashing the FC.  I am seeing messages I do not understand so I need some help getting this resolved.

I am using a Mateksys f204-WMN and Inav 6 loaded to an iMac. 

When i go to the FC page i first see a message:  Connected to: /dev/tty.Bluetooth-Incoming-Port

Then it changes to :    MSP connection successfully closed

I select the proper board and version and load firmware (online).  I get message "Loaded Online Firmware: (545426 bytes)"

Then I plug in the FC.  Press the Flash button and get this message:  Connected to: /dev/tty.Bluetooth-Incoming-Port.

Then nothing.  The bluetooth seems wrong but I am stuck.  Likely an installation issue but well over my head.

Help.

JekleFPV, 2023-02-18:
You got quite a lot further than I did then. On mine it won't even start. Icon bounces for a second and then the dot underneath disappears again...

RobinatiNav, 2023-02-24:
I have iMac and MackBook Pro both running OSX Monterey and I have no problem with iNav 6. My FC is Matek H743-WLITE.

[image]
[image]
[image]

mmosca, 2023-03-02:
> You got quite a lot further than I did then. On mine it won't even start. Icon bounces for a second and then the dot underneath disappears again...

Right click on the configurator and select open. I think some security settings prevent "downloaded" apps do launch the first time.

RobinatiNav, 2023-03-03:
Yes, you need to allow this 3rd party app to be running in your OSX

Kamdozer, 2023-03-26:
Similar issue on Mac with latest OS. Downloaded Configurator 6.0, managed to flash the Inav 6.0 official version from 2.6, after unplugging all peripherals due to problem going in DFU. After successful flash can not connect - just a message MSP port opened successful and after that closed. Led stay red all the time, never changes to blue. No idea what to do now...

closed 2024-06-28 by DzikuVx