iNavFlight/inav-configurator #1711: OSD Setup not possible with DJI Air Unit and Googles 2

boomerschnecke, 2023-02-04:
How to setup the OSD with the DJI RunCam Split Unit with DJI FPV Air Unit and Googles 2?
With INAV 5.1 seems to be not possible?
CLI command set osd_displayport_device = MSP not working
For Matek F405 WMN no firmware for INAV 6 is available
What to do?

MrD-RC, 2023-02-04:
It is not possible with INAV 5.1. That CLI command is for BetaFlight.

closed 2023-02-04 by MrD-RC

boomerschnecke, 2023-02-04:
Hello Darren, could you please give me some hints why it doesnt work? I think everything is setted up like shown in many videos.  I've learned, that the runcam video unit could not be handled with INAV. But are there any restrictions to the DJI Air Unit? Its strange why it doesnt work

MrD-RC, 2023-02-04:
It doesn't work because DJI are shit and didn't implement MSP DisplayPort properly. They added a cut down version with only the BetaFlight font installed on the goggles. If DJI had done things properly, there would not be an issue. 

There is a BetaFlight compatible mode in INAV 6.0 RC1. This will allow it to work with some OSD elements. But the entire INAV OSD won't be supported unless DJI fix their side.

boomerschnecke, 2023-02-05:
Hello Daren, thanks for detailed information. Due to the fact, that there is work around within INAV 6.0 RC1, is there a possibility to transfer the FC firmware Matek F405 WMN to an INAV 6 compatible mode? up to now only INAV 5 compatible SW is availabel. 
 Is there a way to make new SW by my own? Any feedback would be highly appreciated. Thank you
Best regards
Markus

MrD-RC, 2023-02-05:
No. It is a feature for 6.0. If you are that desperate for it you can use the 6.0 RC1. It is pretty stable with no reported major bugs, if any.

boomerschnecke, 2023-02-05:
Thanks Darren, but I need a compatible FC firmware for it, and Matek 405wmn is only having INAV 5 compatible firmware. How to transform this to INAV 6 compatible one?

MrD-RC, 2023-02-05:
The MATEKF405-TE target for that flight controller is released as part of 6.0 RC1. You just need to install it in Configurator 6.0 RC1.

boomerschnecke, 2023-02-05:
But I need to flash with the correct 405 firmware beforehand and sorry, but this in not available up to now for me

boomerschnecke, 2023-02-05:
ok Darren thank you. it seems to work now

boomerschnecke, 2023-02-05:
so INAV 6.0 RC1 works with Matek 405 wmn, but OSD is not visible up to now. Having tried to use UART3 with MSP Displayport. and DJI Air Unit. But no success. What do I do wrong?

MrD-RC, 2023-02-05:
Have you set the video type in the OSD page to BFCompat?

boomerschnecke, 2023-02-05:
yes I did: cli Status tells me ERROR: Invalid setting: osd_video_system

boomerschnecke, 2023-02-05:
and goggles screen now is black

boomerschnecke, 2023-02-05:
should I change from UART3 back to UART5?

boomerschnecke, 2023-02-05:
also in the setup page now "settings validated" is red

boomerschnecke, 2023-02-05:
i changed now back to UART5, at the first setup page still invalid settings an cli status osd invalid

MrD-RC, 2023-02-05:
Don't type it in to the CLI. Use the OSD page.

boomerschnecke, 2023-02-05:
up to now OSD is not working
What can i try in addition?

boomerschnecke, 2023-02-05:
if I select  DJI VTX at Ports than VTX can not be expected, with MSP Display Port OSD is not recognized

MrD-RC, 2023-02-05:
The peripheral DJI VTX on the ports page is for the old v1 and v2 systems. 

The peripheral MSP DisplayPort on the ports page should be used for DJI O3. Then select BFCompat in the video type drop down on the OSD page.

boomerschnecke, 2023-02-06:
Hallo Darren, thanks for your support and clarification of items
Please be so kind an tell me, to which type the RunCam Split Unit belongs? Attached the link from banggood

https://www.banggood.com/RunCam-Link-Split-HD-Air-Unit-2_7K-at-60fps-5_8Ghz-DVR-Record-GyroFlow-Stabilization-WDR-for-DJI-Digital-Vista-Goggles-p-1923686.html?rmmds=myorder&cur_warehouse=CN

Any feedback would be highly appreciated.
At goggles2 its shown as DJI Air Unit (but not the O3!)
Does it belong to old v1  and v2 systems? What to select within Ports configuration?

Also RunCam told me that the Split Unit should work with INAV directly. If I use here the UART3 which type of configuration of the port do I have to use?

I am completely new to this fpv and therefore please apologize the amount of questions

Thank you
Best regards
Markus

MrD-RC, 2023-02-06:
Sorry, I have no idea. I don't use DJI.