iNavFlight/inav-configurator #1714: INAV setup for Goggles 2 and both Air Unit and Air Unit O3 is the same:

boomerschnecke, 2023-02-06:
INAV setup for Goggles 2 and both Air Unit and Air Unit O3 is the same:

1. you need INAV 6
2. Set port as MSP Displayport
3. Ensure OSD feature in Configuration tab is enabled
4. in OSD choose `BF43COMPAT` as video system
5. Power cycle

There are no extra steps or configurations. INAV 6 is available for all flight controllers when `Show Ustable Releases` is selected

_Originally posted by @DzikuVx in https://github.com/iNavFlight/inav-configurator/issues/1711#issuecomment-1418726067_

boomerschnecke, 2023-02-06:
Hallo Pavel, thank you for your reply.
All of this I did several times, but unfortunatelly it doesn't work

DzikuVx, 2023-02-06:
I understand, but I also did it a few time in a last few days and it's working exactly like expected. You also need the latest firmware on both Air Units and Goggles

closed 2023-02-06 by DzikuVx

boomerschnecke, 2023-02-06:
Thank you Pawel for your information:
Could you please check your firmware SW?
I have for Air Unit 01.01.0000
For Goggles: 01.04.00.00
I reflashed now the FC, I updated every SW from Air Unit, Split Unit, Goggles. I checked everything minimum five times.
But no success

boomerschnecke, 2023-02-06:
Hallo Pawel, thank you for your support. The issue is solved now, it works.
Reason: Up to now I only have taken stable released FC Version 5.1.
With the unstable released version 6.x it works now.
Are there any risks with an unstable version ? Or why are they named "unstable"
Best regards 
Markus