iNavFlight/inav-configurator #1717: iNAV 6.0.0  RC2

NorthIceMan, 2023-02-10:
When connect to FC iNAV 6-RC2 goes into CLI.  If try to Exit - it reboots and goes back to CLI

MrD-RC, 2023-02-10:
Which version of the firmware is on the flight controller?

NorthIceMan, 2023-02-10:
Firmware is 5.1

NorthIceMan, 2023-02-10:
Gives "FC Firmware Not Connected" in upper left corner under the iNAV Logo.  Ran it on my new H743 WLITE and on my functioning H743 WING (two different USB devices).  iNAV 6.0.0 RC1 works fine.  Cannot get out of CLI

MrD-RC, 2023-02-11:
That is expected behavior. The firmware version (major and minor) must match the configurator version. 5.1 will only allow access to CLI. Once the flight controller is flashed to 6.0, it will work with the 6.0 Configurator.

closed 2023-02-11 by MrD-RC

NorthIceMan, 2023-02-11:
Yes-I read it clearly that it ONLY works with iNAV 6 but not mentioning the .HEX file (thinking they meant only any previous settings) - I am a retired software dev. Diapers are gone.  Searched everywhere (I think) for the .HEX file but could not find. Funny-RC1 works fine.  I looked at "Click "Load Firmware (Online)" button" today again (in GitHub) and now see that you can not just download the .HEX file (like I did with 5.1) but it must be loaded Online.  Thanks!

NorthIceMan, 2023-02-11:
Some of us young-lads fly into remote areas (go into the wilderness on planes & boats) and if we don't bring a .HEX file with us (in a no internet area) then we are stainless-steel-screwed! ;-)  Funny tho how in an internet based era its the travellers who sometimes are vexed.
Is there anyplace to download the .HEX / Binaries ?

MrD-RC, 2023-02-11:
This is the first time you mentioned the .hex file. That's not related to 5.1 only being given CLI access in 6.0. 

All the hex files can be downloaded from the [firmware release](https://github.com/iNavFlight/inav/releases/tag/6.0.0-RC2) as normal. They are in the assets section at the bottom. The **Load Firmware (Online)** button has only ever downloaded the selected file to configurator. It has never been a _Save As.._ button.

NorthIceMan, 2023-02-11:
AGree 100% - however I first downloaded the hex file from Matek site for H743 (last Dec) under their "Firmwares" button.  I realize that the new ver needs the new hex binaries (its written all over every page) but didn't find it under any search I did in GitHub for "firmware".  I looked in GitHub for a while.  Then (instead of bothering you busy guys AGAIN) decided to try the 5.1 ver.  Maybe I needed another keyword to find it.  It seemed ok for RC1 but not for RC2 of configurator.   My fault-I need to "skate" better in GitHub!
Thanks for the help!