iNavFlight/inav-configurator #1721: WP "user actions" breaks waypoint navigation

ABLomas, 2023-02-13:
Short video demonstrating issue:
https://user-images.githubusercontent.com/8989151/218552591-65620b35-67e2-451c-ae52-334931ab7444.mp4
Text version - mission waypoint with user action set has it's altitude set to MSL+preset altitude, while AGL setting is selected.
This was registered as issue on another branch but it looks like firmware has nothing to do with it:
https://github.com/iNavFlight/inav/issues/8798

In short - looks like INAV-configurator incorrectly parses bit0 in P3 field and and this is reason why alt is set to wrong value.

breadoven, 2023-02-13:
Configurator needs updating to read the WP P3 parameter as a Bitfield rather than a boolean as it does at the moment.

MrD-RC, 2023-02-13:
Configurator has been updated to read P3 as a bitfield. I thought I had tested that everything was toggled correctly. But something must have slipped through.

breadoven, 2023-02-13:
OK missed that change. I was going to do it but seems you've saved me the trouble.

So just updated Configurator and checked this and it appears that the Bit is being set but not unset, so once it's been set it gets stuck true.

breadoven, 2023-02-14:
The other problem is P3 is used as a boolean in various places which affects the altitude calculation depending on whether SL or relative altitude datum is used. Bit 0 needs checking rather than just the boolean state of the P3 value. I've made some changes that have fixed the obvious things, need to check other stuff isn't affected.

breadoven, 2023-02-14:
Do you want me to fix this @MrD-RC given it's related to the MSL Altitude stuff I did originally ?

MrD-RC, 2023-02-14:
@breadoven I found somewhere that was using P3 as a Boolean. So I've sorted that and fixed some other things too. The _Elevation (m)_ and _Grd Dist (m)_ seem to be working correctly to me. But, the elevation profile doesn't seem right. I couldn't see any references to P3 in there. Your thoughts? It may be fine though. It looks the same in 5.1 Configurator.

WP1
[image]

WP2
[image]

WP3
[image]

WP4
[image]

WP5
[image]

WP6
[image]

[test mission.mission.txt](https://github.com/iNavFlight/inav-configurator/files/10735841/test.mission.mission.txt)

MrD-RC, 2023-02-14:
Added a PR https://github.com/iNavFlight/inav-configurator/pull/1722

MrD-RC, 2023-02-14:
There is something causing issues in the elevation profile. If I disable the user actions, it looks as I would expect.

MrD-RC, 2023-02-14:
Got it. It was hidden in `WaypointCollection`.

WP2 now for comparison
[image]

PR #1722 (2023-02-15): Mission Control: Bug fixes related to P3 field
Fixes #1721 

- Added const to make bitfield stuff more readable
- changed `checkAltElevationSanity()` as it was treating P3 as a Boolean. Some times this function was called with explicitly not using absolute altitude. This is still possible. The function can accept a Boolean _or_ the P3 bitfield.
- Fixed issue where alt and user action checkboxes could not be unset.
- Added **.mission** filter to the load mission from file dialog box.
closed 2023-02-16 by breadoven