iNavFlight/inav-configurator #1723: RunCam Split Unit: How to activate Video Unit SD Card writing via Modes

boomerschnecke, 2023-02-15:
Having RunCam Split Unit and INAV 6.0.0 RC2
How to initiate the Video Unit to start recording to  SD Card?
Which Mode? Is there a mode existing to start and stop recording?
What do I have to configurate at Ports? RunCamDevice or DJI FPV VTX? Which speed?
Sorry for asking, but this is absolutely unclear ?

Jetrell, 2023-02-18:
Select ports peripheral `RunCam Device`.  Then the _FPV camera modes_ will appear. Choose `Camera Power.`

boomerschnecke, 2023-02-18:
Thanks for your reply Jetrell
But unfortunately this is not working?
The only thing it starts recording is to be activated via the button or to select immediate automatic recording after power up
With the modes I´m not successful
What do I do wrong?

Jetrell, 2023-02-19:
I have an old Split V2 laying around. And I gave it a test with 6.0RC2.

Connection are :
`VCC`
`GND`
`Video`
`RX` to FC TX
`TX` to FC RX

https://support.runcam.com/hc/en-us/articles/115011438788-INAVFlight
And it functioned correctly.

Something to be aware of.. Once you enter the Port and Mode changes in the Configurator. You need to disconnect the USB and battery power from the FC, to complete a power down reboot. Then it should work once powered again.

boomerschnecke, 2023-02-19:
Hallo Jetrell, thank you for your reply.
you mentioned the "video" pin to be connected? I have not  found any video connection point at the split unit.
I've bought this one:
https://www.banggood.com/RunCam-Link-Split-HD-Air-Unit-2_7K-at-60fps-5_8Ghz-DVR-Record-GyroFlow-Stabilization-WDR-for-DJI-Digital-Vista-Goggles-p-1923686.html?cur_warehouse=CN
Maybe we are talking about different units?
And the other pins I've connected like discribed. So Rx to FC Tx and Tx to FC Rx. But up to now no success...

boomerschnecke, 2023-02-19:
do I have to set anything in addition within configurations?

Jetrell, 2023-02-19:
I see.. That's the Runcam Link digital VTX, with the DVR recorder stack.

There are two ways to go about setting it up in the ports. 
If you are just using the V2 Goggles and the Runcam Link with their stock firmware.. Then select `DJI FPV VTX`.
But if you use WTFOS to gain root access to the DJI goggles and the Runcam Link..  Then select `MSP Displayport`.

And to answer your first question.. INAV or any other flight software that I'm aware of, can not control the Starting and Stopping of the DVR in a digital system, via a Mode selection.. This feature is only done within the goggle firmware.

Jetrell, 2023-02-19:
I see.. That's the Runcam Link digital VTX, with the DVR recorder stack.

There are two ways to go about setting it up in the ports. 
If you are just using the V2 Goggles and the Runcam Link with their stock firmware.. Then select `DJI FPV VTX`.
But if you use WTFOS to gain root access to the DJI goggles and the Runcam Link..  Then select `MSP Displayport`.

boomerschnecke, 2023-02-19:
Hi Jetrell, thanks for your reply.
This is the way to setup OSD to Goggles 2. That's running in the meantime via MSP Displayport
But activation of the DVR recorder stack (is this the unit behind the camera ?) this contains a SD Card in it. And I want to start recording to this SD card via modes? Or is this not possible. For me up to now it seems, that this function is not supported? Or do I understand something compeletly wrong up to now?

boomerschnecke, 2023-02-19:
also  mode "Camera Power" does not exist. Having CAMSTAB and CAMERA CONTROL1-3

Jetrell, 2023-02-19:
>Or is this not possible. For me up to now it seems, that this function is not supported? Or do I understand something compeletly wrong up to now?

Yes. The the unit with the card is the DVR recorder.

I can't remember Runcam adding any modifications to the original INAV analogue Split HD control code. To support the Link Split HD.
Runcam must have made changes to the Betaflight , to accommodate this unit.. 
You will have to ask them Firmily. .. _If they are sure they have made the software changes to suit this unit in INAV_.

boomerschnecke, 2023-02-19:
I've asked the runcam support and they said INAV is supported 
So I thought it should work?

Jetrell, 2023-02-19:
It would be interesting to know, when the support was added? And what was the Pull Request number?

You're the first INAV user who has mentioned this unit . Either here or on the INAV Discord server.
I don't think you're setting it up incorrectly.. More likely, Runcam has not added supported for INAV.

boomerschnecke, 2023-02-19:
could be
I've asked runcam support round about 2 weeks ago
And they said it should work, so the DVR Unit in combination with INAV

Jetrell, 2023-02-19:
The other option would be, if they've done a hack job of it. 
Instead of integrating the MSPOSD and camera control into the same UART... They may expect the user to use two UART's.  

[image]

boomerschnecke, 2023-02-19:
Hello Jetrell, that's exactly what I've done in INAV. Only difference is, that for RunCamDevice I use UART3 and I use UART5 for  DJI FPV but with MSP Displayport for Goggles2

Jetrell, 2023-02-19:
> also mode "Camera Power" does not exist. Having CAMSTAB and CAMERA CONTROL1-3

The camera labels under the modes tab, should change. Once Runcam Device is selected in the ports. 

[image]

boomerschnecke, 2023-02-19:
yes Jetrell correct, sorry was my fault. I've tried so many things.............
But also with this condition it doesn't work

MrD-RC, 2023-04-27:
This seems to be working now, so closing issue.

closed 2023-04-27 by MrD-RC