iNavFlight/inav-configurator #1730: weird behaviour when Tx signal is lost on the ground

RobinatiNav, 2023-02-20:
When on the ground and Armed if Tx is lost the flight mode is switched from Manual to Angle and servos/control surfaces act weird. See attached picture, please.

[image]
[image]
[image]
[image]

MrD-RC, 2023-02-20:
This is an issue with your failsafe setup on the receiver. At no point in these screenshots does INAV think it is in failsafe. The parachute failsafe indicator is never red. It looks like you have your receiver failsafe set up so that the channels go to certain positions.

RobinatiNav, 2023-02-20:
I have failsafe disabled in Rx.

MrD-RC, 2023-02-20:
Then when you lose RC link, the receiver will do whatever the hell it wants. You can't expect this to work correctly if the failsafe is not correctly set up.

RobinatiNav, 2023-02-20:
This is crazy. I tried 3 different Jeti Rx with failsafe disabled and got the same result. I'm flash and clear FC and make new configuration and then come back

MrD-RC, 2023-02-20:
Don't disable failsafe. That is the problem. I don't know what the correct failsafe setting is on Jeti. On FrSky is is no pulses or hold. On Crossfire it is cut. So if you have any options like that, they may be worth trying.

What you need to happen is when you switch off your transmitter, the little parachute top centre should turn red. When you turn the transmitter back on, it should return to grey.

MrD-RC, 2023-02-20:
@DzikuVx do you know what setting Jeti should use for failsafe?

RobinatiNav, 2023-02-20:
Thanks Darren, I am getting crazy. I tried HOLD, Out OFF, and Fail-Safe (particular value for each servo output) but none of those worked :(

DzikuVx, 2023-02-20:
No, I no longer own Jeti. Probably Out OFF should be correctly detectable by INAV

RobinatiNav, 2023-02-20:
OK, what should I see when TX signal is lost apart from RED parachute? What flight mode? I have set RHT in failsafe settings but RTH would not work on the ground.... I am completely gone here. I do not understand that

MrD-RC, 2023-02-20:
The red parachute is the best tell tale on the ground. In flight if the failsafe setting is RTH, that's what it can do. There was a procedure for testing on the ground. But in all honesty, I just use the parachute these days. 

I believe the procedure was something like:
1. Prop off
2. Arm
3. Raise the throttle 
4. Run with the model and transmitter for 50 metres
5. Switch off the transmitter 

Then you should see the motor increase and the elevator pitch up.

MrD-RC, 2023-02-20:
@RobinatiNav just out of interest. What happens on the receiver page? Could you make a screen grab with and without RC link connected.

It sounds like failsafe may be working, going by !FS! in the OSD. Emergency landing would also be correct.

RobinatiNav, 2023-02-20:
it remains the same

[image]

RobinatiNav, 2023-02-20:
Is it possible to get nav modes active on the ground? just for testing. I managed it somehow and it looked like it woked. The rad parachute was there and throttle at cruse speed, pitch as up. I do not know ho I did it

RobinatiNav, 2023-02-20:
I unplugged the Rx from the FC to know if it is for sure the RX failsafe doing problems but it behaved the same like when only Tx signal was lost. The manual mode was switched to the angle mode but no red parachute.

RobinatiNav, 2023-02-20:
I tried something. I went aprox. 10m away from my iMac to arm the FC from some distance, then came back and simulated the Tx signal loss. See result on pictures: 

[image]

[image]

RobinatiNav, 2023-02-20:
I have one important question at the end. When failsafe (RTH) is activated two modes are active simultaneously. RTH and ANGLE. In case the GPS signal is not present for some reason RTH will not work but ANGLE will do like when testing on the table. In this scenario the throttle will by cut off, pitch and roll will act weird for no reason, and plane will go down almost immediately. Why is it working like that?

RobinatiNav, 2023-02-21:
Hi Darren,

could you please check weird behaviour when Tx signal is lost on the ground <https://github.com/iNavFlight/inav-configurator/issues/1730> my last comment. I think I found a bug. Thank you

Robert.

RobinatiNav, 2023-02-21:
After huge investigation I can say the angle mode I don't see it is doing what I expect from it is probably not a fault but performed emergency landing. I do not like it goes down in spiral but I can confirm it is safe. Tell me if I am wrong.

closed 2023-02-23 by RobinatiNav