iNavFlight/inav-configurator #1731: Configurator LED's will not save if any orientations are selected in  6.1 & 6.2

Helipro, 2023-02-23:
Current Behavior
When changing and saving leds in the configurator in 6.1 & 6.2 and any orientations are selected it will not save.  Orientations will be loaded but must be deleted to save any new led configuration in 6.1 & 6.2.
Tested on matek f411 and 2 x f765 all with this new issue that didnt occur in previous versions.

There was a new addition of the strobe effect that may have caused this issue.
To fix the issue I deleted all the orientations and was able to save my new different led configuration.

MrD-RC, 2023-02-23:
You mean 6.0 RC1 and RC2 I take it. Please be accurate with version numbers. 

This is the only change to the LED setup since 5.1 https://github.com/iNavFlight/inav-configurator/pull/1647 @mmosca maybe you could take a look?

mmosca, 2023-02-23:
I can have a look next week, as I am currently out of town.

mmosca, 2023-03-01:
Just a quick update. 
Had a quick look today. I can reproduce the issue and I can also confirm that setting orientation on the cli works, so at least there is a workaround.

I will continue the investigation and focus on the configurator.

mmosca, 2023-03-02:
The fix has been merged on the release 6.0.0 branch and should be included either in the next configurator RC or final release.

closed 2023-03-02 by mmosca