iNavFlight/inav-configurator #1737: Stuck in CLI

jerrymou, 2023-03-04:
After flashing new firmware for my Flywoo Firefly hex nano successfuly, I am stuck in CLI with this message:
You need to upgrade your firmware before you can use the Setup tab. I can't do anything else! Have tried several times. Please help

jerrymou

MrD-RC, 2023-03-04:
You need to use the same firmware and configurator major and minor versions.

closed 2023-03-04 by MrD-RC

jerrymou, 2023-03-04:
My Flywoo Firefly Hex Nano INAV has this on the label on the box:
Firmware INAV 3.0.0 Flywoo F745 HEX
I have downloaded and installed INAV 3.0.0 
In INAV 3.0.0 the only 2 firmware available to me are: FLYWOOF745 or FLYWOOF745NANO. There is no HEX version available as per label on my Flywoo box. I have tried both of them with the latest stable release and the same thing happens.