iNavFlight/inav-configurator #1748: 6.0 Mac Dmg Installs v 5.1?

peterbrady, 2023-03-29:
I downloaded the 6.0 release from the releases page: and received INAV-Configuration_macOS_6.0.0.dmg which has 132,946,219 bytes. After installing and running it reports that its v5.1

RobinatiNav, 2023-03-29:
I installed the same and it looks like this:

[image]

RobinatiNav, 2023-03-29:
> I installed the same and it looks like this:

Actually, it is not installing but drag and drop.
> 
> [image]

peterbrady, 2023-03-30:
Apologies. I did have a previous version, 5.1 installed. Not sure why the DMG installer did not overwrite for me, but it works after deleting the old version. Closing now.

closed 2023-03-30 by peterbrady