iNavFlight/inav-configurator #1752: GPS not shown in the GPS page with a telemetry communication with INAV Configurator 6.0

wdliu356, 2023-04-04:
Hi, I'm building a 4 inch quad with MatekH743 flight controller board, mRo GPS u-Blox Neo-M8N IST8308 as my GPS and magnetometer. I'm also using RFD 900x-US as my telemetry radio for remote communication with the INAV configurator.
GPS shows well in the Setup page as is shown in the following figure.

[image]

But on the GPS page and the Mission Control Page, the drone cannot localize itself on the map, shown in the following two figures.

[image]
[image]

The ports page is shown as follows:
[image]

Everything is god when I'm using a wired communication between the my drone and laptop but wireless one is just not working that well. I checked that there was a similar issue posted in the inav repo and collaborators asked that person to repost that to the configurator repo but I can't find the repost so I post another issue here.

OptimusTi, 2023-04-05:
Your telemetry radio too "loud" for the GPS?

OptimusTi, 2023-04-07:
Oh, I didn't notice the 3d fix. Are you using an rx system that may limit telemetry sync like exprlrs or something like that? Check the telemetry settings and try UBLOX7

OptimusTi, 2023-04-07:
Oh, I didn't notice the 3d fix. Are you using an rx system thatt may limit telemetry readings like exprlrs or something like that?

wdliu356, 2023-04-07:
Which telemetry do you mean? Do you mean the telemetry connection with the ground station or the one with a remote controller? For that with the remote controller, yes, I'm using an Rx system. Will that affect that much? Since even if I didn't connect with the remote controller, it still doesn't show the GPS stuff

OptimusTi, 2023-04-07:
Which rx system?

wdliu356, 2023-04-07:
Just wanna double check when you are talking about rx system, you mean communicating by rx and tx ports right?

MrD-RC, 2023-04-07:
He means the RF link. The transmitter and receiver you are using for the RC communication.

wdliu356, 2023-04-11:
The telementry radio for remote controller I'm using is TBS crossfire nano rx. Any clue for that?

OptimusTi, 2023-04-11:
Yeah I have the same setup and it's working. I'm on 7.0 but it should be the same. I'm going to get install 6.0 and let you know

OptimusTi, 2023-04-12:
Confirmed. Works in 6.0. I'd reinstall your configurator and check wiring.
[image]

wdliu356, 2023-04-12:
Do you have any idea about what kind of wiring problem can lead to such a problem? I tried both INav 5 and 6 and both of them share the same problem so I would think it might be the wiring issue. Am I supposed to focus on the wiring of the telemetry radio?

OptimusTi, 2023-04-12:
At this point, I'd reinstall the configurator. Looks like the GPS is communicating with the FC.

wdliu356, 2023-04-13:
And another interesting is that though it shows no gps fix on the GPS page and no point on the current location, when I click the 'center', it centers on the correct place. When I zoom in the map, it shows the correct area I'm currently at. But still not showing GPS fix

OptimusTi, 2023-04-13:
> May I check what system are you working on? Windows or Linux? I'm working at Windows system and tried both the .exe file and the zip file of the release. But it still seems not showing on the GPS page

I'm on latest Windows 10. I installed 6.0 using the Windows installer INAV-Configurator_win64_6.0.0.exe

wdliu356, 2023-04-13:
> And another interesting is that though it shows no gps fix on the GPS page and no point on the current location, when I click the 'center', it centers on the correct place. When I zoom in the map, it shows the correct area I'm currently at. But still not showing GPS fix

I also tried that on a linux machine and it shares the same problem. 
It can locate itself but it will not show its position on the map and shows no 3d fix on the GPS page.

wdliu356, 2023-04-13:
Another thing is that, with the current setup, I lost the communication with the drone after I arm it

wdliu356, 2023-04-13:
I've actually done this several times but nothing changed. I believe I'm not the only one meeting this issue.
https://github.com/iNavFlight/inav/issues/4553
The link is another person sharing the same issue with me. This might be a hidden bug that has been there for many years maybe?

wdliu356, 2023-04-13:
Hey, I just fixed issue by not clicking the wireless mode when connecting with the drone, though I'm doing that through telemetry. Maybe you need to double check that button or record that in the readme.

OptimusTi, 2023-04-13:
I'm still not having the issue even if I select wireless mode. Works fine here.