iNavFlight/inav-configurator #1757: SITL: save all settings of that section

robustini, 2023-04-14:
It would be helpful if the motors and servos order could be saved in the SITL profiles, every time I go back to it it goes back to the default and I have to change it manually.
This for example the correct order for the model I use, and the default is another.
"Serial to TCP" also always comes back enabled but I don't use it.

[image]

Scavanger, 2023-04-14:
I have just tested the latest version (Branch release_6.1.0): Works as intended.

You have to create a new profile and save the profile after the changes. The default profiles cannot be changed, a message that they cannot be overwritten should be added here.

robustini, 2023-04-15:
> I have just tested the latest version (Branch release_6.1.0): Works as intended.
> 
> You have to create a new profile and save the profile after the changes. The default profiles cannot be changed, a message that they cannot be overwritten should be added here.

Thanks!
Exactly, what is missing is a warning that the two default profiles are not writable.

closed 2023-04-15 by robustini