iNavFlight/inav-configurator #1759: With Acro Mode the plane drifts down with an angle around 45degrees

boomerschnecke, 2023-04-16:
When Acro Mode is activated ( and this is the standard mode) you are not able to start the plane or fly with this mode.
The plane is drifting dramatically down with an angle around 45 degrees.
Seems the the elevator stick is pulled down extremly

With other modes you can fly, but not with acro mode

INAV 6.0 with Matek FC405TE and latest SW

rts18, 2023-04-17:
Are you aware Acro Mode is not a LEVEL stabilized flight mode, as Angle and Horizon modes?
Acro Mode only provides attitude stabilization, with its strength being determined by the axis i-term gain.

In short, Acro mode is a dampened version of Manual mode, for fixed-wing pilots that prefer a smoother stick feel, when performing aerobatic flight.

boomerschnecke, 2023-04-17:
you are correct, but with this kind of mode its not possible to fly the plane.
If you are switching to this mode, it always starts directly to fly down to the earth with around 45 degrees.
Even with the full activation of the elevator switch up, its not able to fly straight ahead, it always is flying  down
This happens also with the manual mode.
So with manual mode not able to fly, with other stabilized modes you can fly
This is strange I think

MrD-RC, 2023-04-17:
If the problem is also in manual mode. The issue is with the physical setup of the plane. Hands off, it should fly reasonably straight in manual. 

I would check the CG and your control surfaces. Maybe check the outputs page in Configurator and the servo midpoints (if you have continuous servo trim enabled). This may also give clues. 

Rule of thumb. If it flies badly in manual. It is the plane. It would fly the same without the flight controller installed.