iNavFlight/inav-configurator #1769: No PPM option found in INAV 6 configurator for operating Flysky FS-i6S with FS-IA10B receiver

Pratham340, 2023-04-28:
Connected PPM single wire to Mateksys F722-WING (RX2) worked just fine with Betaflight but when tried with INAV there was no option for PPM and just wasn't working. Any help from anybody would be highly appreciated! :)
[image]
[image]

stronnag, 2023-04-28:
By design. PPM was removed from the firmware some time ago.

Pratham340, 2023-04-28:
Hi Stronnag,
I am new to this and I have started putting my hands into this recently.
Then how do people with FS-i6s or i6x or i6 use INAV for their multi-copters or fixed wings?

stronnag, 2023-04-28:
One of:
* IBUS
* Use an older version of INAV
* Upgrade to more modern RX/TX

Pratham340, 2023-04-28:
How do we use IBUS?

Pratham340, 2023-04-28:
Thanks, I got it.

closed 2023-04-28 by Pratham340