iNavFlight/inav-configurator #1771: Does the INAV firmware version affect the communication range.

al1965al, 2023-04-29:
Question
Can the communication range change with different INAV versions but with the same receiver firmware version?
Let me explain.
If INAV 3.0.2 is installed on the wing, I see RSSI -100% in the configurator.
I can move around the apartment - and RSSI is 100% stable.
But if I install INAV 6.0 - with the same receiver firmware,the same CRSF protocol -  in the configurator I see RSSI 100....95..97and changes already in half a meter..
 The values ​​are not stable and change. It looks like the connection is worse than in version 3.0.2.
Therefore, the question is.
 Or can the quality of the connection depend on the version of the iNAV firmware? Or are it is just visible values ​​from developers that DO NOT affect really the quality of communication.
Thanks

MrD-RC, 2023-04-29:
The INAV firmware has no effect on the signal strength of the RF system. I believe between those versions there was a change to how LQ was displayed.

closed 2023-04-29 by MrD-RC

al1965al, 2023-04-29:
But this is a fact.
In version 6, at a distance of 30 cm from the transmitter, the RSSI value is already changing.(I check at power -10mW)
And in 3.0.2 - the value is very stable 100%. Maybe the developers really installed a different algorithm for RSSI readings and the communication quality remained the same.

OptimusTi, 2023-04-29:
If you use the CRSF protocol, RSSI in % uses the signal strength for reference. The 100% changes when the signal strength drops below the value of "osd_rssi_dbm_max". The default value is -30dBm. 

If the number drops at half a meter, that tells me that there is something wrong with your equipment. Check your antennas and RX.

[image]

al1965al, 2023-04-29:
I don't want to offend you. BUT - why does everyone put forward some ideas - without even checking.
Check it out yourself. Install version 3.0.2.Power 10mW.
Look at the RSSI values ​​​​in the configurator - it will be 100. Stable! Even increasing the distance.
Then install 6.0
The RSSI values ​​in the configurator immediately change..As if they are more sensitive. Increasing the distance immediately reduces the RSSI value.
The equipment is working properly! At 3.0.2 at 10mW I flew 15km!
And I fly not the first day.

OptimusTi, 2023-04-29:
Before INAV 4.1, RSSI% was LQ. You can go back to that if you prefer.

al1965al, 2023-04-29:
Do not confuse the correction of the horizon in version 6 and RSSI. I already flew only on 3.0.2.
But I decided to switch to 6 because of the corrected horizon.
I repeat - it is possible that the range will not change, and these are just the readings in numbers. But the visible difference in the configurator is very large.

al1965al, 2023-04-29:
_-Before INAV 4.1, RSSI% was LQ.-___
And now in version 6 - what is RSSI?
Could this be the reason for the different readings?

OptimusTi, 2023-04-29:
No worries. I'm not confusing anything.

If you want INAV 6 RSSI% to behave like it did in 3.0.2, assign LQ to a channel and use that.

[image]

al1965al, 2023-04-29:
I got it. Thank you very much for your help!