iNavFlight/inav-configurator #1776: Manual mode: airplane goes straight down

boomerschnecke, 2023-05-10:
Hello all,
once again the issue, that the airplane goes straight down with around 45 degree with manual mode
With other modes it flys. (more or less...)

BUT its pumping like hell if throttle is a higher speed. And Im flying a ZOHD Drift with 2s. Means there is not to much power anyhow.

Whats wrong?
WITHOUT the FC the airplaine flys great.
So there must be any effect by the FC
FC is Matec F405 TE with newest SW and INAV 6.1

MrD-RC, 2023-05-10:
Do you have DVR of this happening? Can you also post screenshots of your mixer and outputs pages.

With manual, the PIDFF loop is completely bypassed. All that INAV does is take your inputs, add any specified expo and rates, applies any logic conditions, puts it through the mixer, then through the outputs.

boomerschnecke, 2023-05-11:
Hi Darren, should I send some screenshots or the diff all file? If screenshots could I send this to you email account directly? Or should / can I upload this here directly?

boomerschnecke, 2023-05-12:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

rsilk1949, 2023-07-25:
Is the plane balanced on the centre of gravity marks with the battery installed. This is critical in manual mode.

MrD-RC, 2023-07-25:
I couldn't see anything in the screen shots. But it has to be a physical/mechanical issue. If manual has issues but the other modes (assuming Angle or Horizon) are flying ok. They are compensating for something in the setup.

As suggested, check CG. I'd also add make sure there is no trim on the transmitter. If you flew it without an FC before, maybe you still have some trim in there. I know this model only needs a little elevator throw. So with some setups, a little accidental trim on the transmitter could cause big problems.

closed 2023-11-03 by DzikuVx