iNavFlight/inav-configurator #1777: inav-configurator 6.0 mixer page show error

rcdrones, 2023-05-12:
Thank you comunity make great fc SW.

today i find a minor bug:
[image]

thanks , Jian

MrD-RC, 2023-05-12:
I don’t see a bug. The servos are where they should be.

rcdrones, 2023-05-16:
S11 is board pin‘s name. 
Servo 9 should in S11 . (actually i use  OSC to confirm is info, S11 have right PWM wave)

-------------
But inav-configurator 6.0 show  Servo 9 in S7 is minor bug.

MrD-RC, 2023-05-16:
That’s not how it works. You cannot specify pin output. The servo index, 9 in this case, only specifies the order of the servos. Not the output pin. INAV decides the output pin and displays it in the output mapping table. Servo 9 should be outputting to signal pin 7. 

What you see on screen is correct.

closed 2023-05-16 by MrD-RC