iNavFlight/inav-configurator #1779: INAV 6.1.0 - RPM installed is broken

stronnag, 2023-05-13:
The `INAV-Configurator_linux64-6.1.0-1.x86_64.rpm` package is not installable on modern Fedora (tested on F38), due to a bogus dependency.

```
$ sudo dnf localinstall /tmp/INAV-Configurator_linux64-6.1.0-1.x86_64.rpm
Last metadata expiration check: 0:28:07 ago on Sat 13 May 2023 12:35:58 BST.
Error: 
 Problem: conflicting requests
  - nothing provides libgconf-2-4 needed by INAV-Configurator_linux64-6.1.0-1.x86_64
```

Neither the configurator, nor its dependencies have any dependency on the obsolescent `libgconf` so this is a packaging error.

notsaeegavas, 2023-07-07:
Having the same issue. Ran via GUI and CLI with dnf -- Output is here:

`Last metadata expiration check: 0:10:08 ago on Thu 06 Jul 2023 06:50:08 PM MDT.
Error: 
 Problem: conflicting requests
  - nothing provides libgconf-2-4 needed by INAV-Configurator_linux64-6.1.0-1.x86_64 from @commandline
(try to add '--skip-broken' to skip uninstallable packages)
`

stronnag, 2023-07-07:
Fixed by #1790

closed 2023-07-07 by stronnag

MiniSentry, 2024-03-27:
This issue still exists in .deb builds, as tested in Debian 12.