iNavFlight/inav-configurator #1785: Inav 6.1 Display Error on Debian 11 & LXQt

jfruitet, 2023-05-23:
I have installed Google Chrome and INAV Configurator on a emachines under  Debian 11 with LXQt as Desktop Environment.

INAV start but the upper and left menue frames are not  correctly displayed.

Is this a known issue.

JF - Nantes - France

[image]

jfruitet, 2023-05-23:
[image]

jfruitet, 2023-05-23:
This is a little better but the upper frame still errorness.

stronnag, 2023-05-23:
This is a problem with your setup / hardware / video drivers.
The INAV-Configuator works perfectly on Debian Stable using any of Gnome, Xfce or LXQT desktops for me.

jfruitet, 2023-05-23:
Thanx.
I'll look that stuff.

jfruitet, 2023-05-23:
Have I to install node.js ?
See "https://github.com/iNavFlight/inav-configurator" 
Building and running INAV Configurator locally (for development or Linux users)

jfruitet, 2023-06-18:
I think is the Graphic Card. To old... and to smal definition.

closed 2023-06-18 by jfruitet