iNavFlight/inav-configurator #1786: MATEK f405 CTR Gyroscope/accelerometer going crazy and spinning in random directions

vortexsilvia, 2023-06-07:
I'm using a Mateksys F405CTR with 6.1.0 Inav FW, and running iNav configurator version 6.0.0. Whenever i connect my fc to iNav, in the setup tab the quad just flips constantly in random directions, sometimes on the roll axis, and sometimes on a random amount of axes. Whenever i try to fly the quad, when any amount of throttle is applied the quad just goes apeshit and starts to randomly spin on the heading axis. I've repeated the accelerometer calibration multiple times and nothing seems to help. 
[image]
The quad is placed on a level surface. I have no idea why any of this is happening since everything worked perfectly with Betaflight.
[image]
[image]
If anyone can help me that would be greatly appreciated. I flashed this exact FW on the fc:
[image]

rsilk1949, 2023-07-25:
Try running the 6 way calibration again.

quetiano, 2023-07-25:
Hello vortexsilvia
My personal experience is that your flight controller has come to an end. The gyro or accelerometer has become faulty, which has happened to me, there is nothing to fix it, but, maybe it might be possible to purchase an external gyro with accel and solder it to the faulty flight controller and get some help with modifying inav to read and use the external gyro with accel, otherwise, purchase another flight controller as they are made to give up sooner or later...think about it...how would Matek remain open for business if all flight controllers were made perfectly and made to last ????

Chilllik, 2023-11-16:
atomrc f405 navi
imu die(finally) after update to 7. all safety points were met.

closed 2025-01-18 by b14ckyy