iNavFlight/inav-configurator #1797: NEUTRONRC F435 AIO Unable to recognize GPS

FlyChenPiPi, 2023-06-29:
I tried
6.1.1- NEUTRONRCF435MINI -2023-6-14 7:31
as well as
6.1.0- NEUTRONRCF435MINI -2023-5-13 17:51
Unable to effectively recognize GPS
I tried two modules, BZGNSS BZ-181 and Beitian BN-180
In addition, changing to UART1 or 2 or 3 has no effect. The Baud is 115200bps and other Baud have been tried.
The TX indicator light of BN-180 will flash in an irregular manner after FC operates normally.
I am sure there is no problem with my hardware connection, as GPS can be correctly recognized in the Betaflight4.3.3 firmware, but replacing it with INAV is not feasible.

我尝试了
6.1.1 - NEUTRONRCF435MINI - 2023-6-14 7:31
以及
6.1.0 - NEUTRONRCF435MINI - 2023-5-13 17:51
均不能有效地识别到GPS
我尝试了BZGNSS BZ-181与Beitian BN-180两款模块
并且更换到UART1或者2或者3都没有效果，波特率115200bps和其他波特率都试过了。
其中BN-180的TX指示灯在FC正常工作后会以一种不规律的方式闪烁。
我肯定我的硬件连接没有问题，因为在Betaflight4.3.3固件中GPS可以被正确的识别，但更换为INAV则不行。

superhappyfun, 2023-09-11:
I'm also having the same issue with this board.  I'm using a Matek M8Q-5883 GPS module.  Usually this happens when I've mixed up the RX and TX pins.  However, my orientation of these pins was correct (I tried both ways just to be sure).  The GPS doesn't get any messages.  Is there a fix for this?

superhappyfun, 2023-09-12:
I was using UART2 and I also tested UART5 using iNav 6.1.1 and neither UART gives me a GPS signal.  I compiled iNav 7.0.0 and flashed it and I was unable to get a GPS signal on that either.  It there an difference between this AIO board and the non-AIO NeutronRC F435 mini?