iNavFlight/inav-configurator #1804: OSD Elements still appearing after being removed using the OSD tab

Painless360, 2023-07-12:
Thanks to all of the developers for all the hard work on making INAV such a fab tool!

Today, I setup a plane with INAV 6.1.1 using a AtomRC F405 Wing flight controller and Walksnail goggles/VTX. I moved all of the elements that I use in the OSD into position and removed all those that were not needed.

Powering the model and goggles the OSD worked great but there were two unexpected OSD elements in the middle of the screen (OSD elements 110, 112?)

I assumed I'd clicked them by mistake but checking in Configurator OSD tab showed that no such OSD element was set to appear. Removing all of the elements from the OSD layout still showed the errant OSD element displayed in the goggles despite configurator showing that the OSD had nothing selected to be shown.

I've never seen this before. I updated the Walksnail goggles and VTX to the latest FW (33.39.10) in case I was seeing an odd MSP issue with mismatched versions ot an odd font issue but it was the same. 

Finally, I copied the OSD section from the Diff All and edited it by hand to set the status of the OSD_LAYOUT table for those unwanted elements to be H. I then copied back this table via the CLI. That fixed it.

HOW IT SHOULD WORK:
Removing all elements from the OSD screen should ensure  that all of the OSD elements are removed from the FC. 

SUGGESTION:
- Investigate why this can happen and fix it
and, ideally 
- Provide a way in Configurator to completely 'wipe' an OSD layout completely and reset all of the elements in that OSD layout to be hidden with default locations for when this happens.

---

- FC Board name and vendor: AtomRC F405 Wing
- INAV version string: INAV 6.1.1

Diff All for reference (this was before the fix, only OSD elements up to

# version
# INAV/ATOMRCF405NAVI 6.1.1 Jun 13 2023 / 10:02:51 (42ced253)
# GCC-10.2.1 20201103 (release)

# start the command batch
batch start

# reset configuration to default settings
defaults noreboot

# resources

# Mixer: motor mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

# Mixer: servo mixer
smix reset

smix 0 1 0 80 0 -1
smix 1 2 0 80 0 -1
smix 2 3 1 80 0 -1

# Outputs [servo]
servo 1 1000 2000 1539 -100
servo 2 1000 2000 1479 -100
servo 3 1000 2000 1530 100
servo 4 1000 2000 1500 -100
servo 6 1000 2000 1500 -100

# safehome

# features
feature MOTOR_STOP
feature GPS
feature PWM_OUTPUT_ENABLE
feature FW_LAUNCH
feature FW_AUTOTRIM

# beeper

# blackbox
blackbox -NAV_ACC
blackbox NAV_POS
blackbox NAV_PID
blackbox MAG
blackbox ACC
blackbox ATTI
blackbox RC_DATA
blackbox RC_COMMAND
blackbox MOTORS
blackbox -GYRO_RAW
blackbox -PEAKS_R
blackbox -PEAKS_P
blackbox -PEAKS_Y

# Receiver: Channel map
map TAER

# Ports
serial 0 0 115200 115200 0 115200
serial 3 2 115200 115200 0 115200
serial 4 33554432 115200 115200 0 115200

# LEDs

# LED color

# LED mode_color

# Modes [aux]
aux 0 0 1 1325 2100
aux 1 2 0 900 1225
aux 2 12 0 1225 1500
aux 3 10 0 1750 2100
aux 4 21 2 1675 2100
aux 5 54 4 1675 2100
aux 6 13 3 1675 2100

# Adjustments [adjrange]

# Receiver rxrange

# temp_sensor

# Mission Control Waypoints [wp]
#wp 0 invalid

# OSD [osd_layout]
osd_layout 0 0 23 0 H
osd_layout 0 1 3 17 V
osd_layout 0 7 46 0 V
osd_layout 0 9 45 4 V
osd_layout 0 11 3 15 V
osd_layout 0 12 23 17 V
osd_layout 0 13 3 9 V
osd_layout 0 14 3 0 V
osd_layout 0 15 44 9 V
osd_layout 0 22 22 0 V
osd_layout 0 23 24 0 V
osd_layout 0 28 44 17 V
osd_layout 0 30 11 15 V
osd_layout 0 31 3 1 V
osd_layout 0 32 3 16 V
osd_layout 0 40 24 2 H
osd_layout 0 41 1 2 H
osd_layout 0 42 1 3 H
osd_layout 0 110 23 11 V
osd_layout 0 111 18 9 H
osd_layout 0 112 23 10 V
osd_layout 0 135 41 16 H

# Programming: logic

# Programming: global variables

# Programming: PID controllers

# master
set gyro_main_lpf_hz = 25
set dynamic_gyro_notch_q = 250
set dynamic_gyro_notch_min_hz = 30
set gyro_zero_y = -2
set ins_gravity_cmss =  997.481
set acc_hardware = BMI270
set acczero_x = -15
set acczero_y = 8
set acczero_z = -11
set accgain_x = 4098
set accgain_y = 4086
set accgain_z = 4100
set align_mag = CW270FLIP
set mag_hardware = NONE
set baro_hardware = SPL06
set motor_pwm_protocol = DSHOT600
set failsafe_procedure = RTH
set align_board_yaw = 1800
set platform_type = AIRPLANE
set has_flaps = ON
set model_preview_type = 14
set servo_pwm_rate = 100
set small_angle = 180
set ahrs_inertia_comp_method = ADAPTIVE
set applied_defaults = 3
set gps_sbas_mode = AUTO
set gps_ublox_use_galileo = ON
set airmode_type = STICK_CENTER_ONCE
set nav_extra_arming_safety = ON
set nav_wp_radius = 5000
set nav_wp_max_safe_distance = 500
set nav_rth_allow_landing = NEVER
set nav_rth_altitude = 7000
set nav_fw_control_smoothness = 2
set nav_fw_launch_velocity = 150
set nav_fw_launch_accel = 1500
set nav_fw_launch_max_angle = 180
set nav_fw_launch_motor_delay = 100
set nav_fw_launch_timeout = 10000
set nav_fw_launch_max_altitude = 5000
set nav_fw_launch_climb_angle = 25
set osd_video_system = AVATAR
set osd_units = IMPERIAL
set osd_time_alarm = 20
set osd_alt_alarm = 110
set osd_home_position_arm_screen = OFF
set osd_switch_indicators_align_left = OFF

# profile
profile 1

set fw_p_pitch = 15
set fw_i_pitch = 5
set fw_d_pitch = 5
set fw_ff_pitch = 80
set fw_p_roll = 15
set fw_i_roll = 3
set fw_d_roll = 7
set fw_p_yaw = 50
set fw_i_yaw = 0
set fw_d_yaw = 20
set fw_ff_yaw = 255
set max_angle_inclination_rll = 450
set dterm_lpf_hz = 10
set fw_turn_assist_pitch_gain =  0.400
set nav_fw_pos_z_p = 25
set nav_fw_pos_z_d = 8
set nav_fw_pos_xy_p = 55
set d_boost_min =  1.000
set d_boost_max =  1.000
set rc_expo = 30
set rc_yaw_expo = 30
set roll_rate = 18
set pitch_rate = 9
set yaw_rate = 3

# profile
profile 2

set fw_p_pitch = 15
set fw_i_pitch = 5
set fw_d_pitch = 5
set fw_ff_pitch = 80
set fw_p_roll = 15
set fw_i_roll = 3
set fw_d_roll = 7
set fw_p_yaw = 50
set fw_i_yaw = 0
set fw_d_yaw = 20
set fw_ff_yaw = 255
set max_angle_inclination_rll = 600
set max_angle_inclination_pit = 600
set dterm_lpf_hz = 10
set fw_turn_assist_pitch_gain =  0.400
set nav_fw_pos_z_p = 25
set nav_fw_pos_z_d = 8
set nav_fw_pos_xy_p = 55
set d_boost_min =  1.000
set d_boost_max =  1.000
set fw_level_pitch_trim =  2.000
set rc_expo = 30
set rc_yaw_expo = 30
set roll_rate = 18
set pitch_rate = 9
set yaw_rate = 3

# profile
profile 3

set fw_p_pitch = 15
set fw_i_pitch = 5
set fw_d_pitch = 5
set fw_ff_pitch = 80
set fw_p_roll = 15
set fw_i_roll = 3
set fw_d_roll = 7
set fw_p_yaw = 50
set fw_i_yaw = 0
set fw_d_yaw = 20
set fw_ff_yaw = 255
set max_angle_inclination_rll = 450
set dterm_lpf_hz = 10
set fw_turn_assist_pitch_gain =  0.400
set nav_fw_pos_z_p = 25
set nav_fw_pos_z_d = 8
set nav_fw_pos_xy_p = 55
set d_boost_min =  1.000
set d_boost_max =  1.000
set rc_expo = 30
set rc_yaw_expo = 30
set roll_rate = 18
set pitch_rate = 9
set yaw_rate = 3

# battery_profile
battery_profile 1

set battery_capacity = 2200
set battery_capacity_warning = 770
set battery_capacity_critical = 770
set throttle_idle =  5.000
set nav_fw_launch_thr = 1900
set nav_fw_launch_idle_thr = 1750

# battery_profile
battery_profile 2

set throttle_idle =  5.000

# battery_profile
battery_profile 3

set throttle_idle =  5.000

# restore original profile selection
profile 2
battery_profile 1

# save configuration
save

#

MrD-RC, 2023-07-12:
@Painless360  Hi Lee,

OSD Element 110 is CRSF LQ
OSD Element 112 is CRSF TX Power

Could it be that you copied the OSD part of the diff from a setup that uses Crossfire or ELRS, to a model that uses another system?

Configurator currently hides elements on the OSD page that you can't use. But the problem is that if a layout is transferred over from another model, but has set-up differences. You can't just switch off the settings that are no longer valid. This has been noticed on airspeed in the past too.

There are a couple of options:
1. Change the OSD page so that is always shows all elements. But has a warning message to say they won't work. This at least allows people to switch things off if unsupported. (Configurator change)
2. Check the OSDLAYOUT when importing for edit. If an element is for a disabled feature, but visible. Automatically hide it. (Configurator change)
3. Change the OSD code, so that elements aren't drawn if the functionality is not enabled. This would just ignore rogue elements, transferred over from other setups. There are `#if defined` blocks in the code (seen for the CRSF elements [here](https://github.com/iNavFlight/inav/blob/588866b6c608915a5518ef1f0f2e30bd17aa589f/src/main/io/osd.c#L2236)). But these elements are still shown. Also, I think it would be better if the draw for these elements is not called at all. (firmware change)

I'm thinking a Configurator change may be better. That way, the element draw won't even be called. But there are a couple of options.

By the way, Lee. This will hide those elements
```
osd_layout 0 110 23 11 H
osd_layout 0 112 23 10 H
```
You just need to change the last character from **V** *visible* to **H** *hidden*. For ages I was confused, and thought the V was vertical and the H was horizontal.

Painless360, 2023-07-15:
Hey Darren. 

Thanks for coming back to me! I hope you're having fun! 

Could this be due to it being supplied with CRSF set as the Rx option, then changing it to SBUS in my build 'hid' those CRSF related OSD elements?

That seem to be plausible. I figured out that **osd_layout 0 110 23 11** H and **osd_layout 0 112 23 10 H** trick after looking at your great OSD layout tool website but I think that should be a way to 'delete' all elements and 'reset to blank' in the Configurator to stop these 'orphaned' OSD elements from being displayed and relying on the pilot to find what needs to be 'on' to enable access to them in the OSD tab to remove.

Is there a table of what the OSD numbers refer to for reference? I can't find one in the documentation...

Thanks again!

Lee

closed 2025-01-18 by b14ckyy