iNavFlight/inav-configurator #1805: Servo outputs shown incorrectly if servo number is skipped

jeffhendrix, 2023-07-13:
When a servo number is skipped, inav (the flight controller fw) will skip outputs, but configurator shows them as not skipped.
Please try this on the bench before closing as not an issue.
This is very easy to reproduce and will show the problem.
Create a simple 4 channel airplane.
[image]
Notice how Servo 1 is on S3 (on the FC), Servo 2 on S4, Servo 3 on S5 and Servo 4 on S6
Add a new mixer rule for servo 5 and notice that it will map it to S7 on the FC.
[image]
Plug a servo into S7 on your FC and verify that it moves when you move the channel.
Now change the servo number to 6 and notice that it shows servo 6 on output S7
[image]
But that channel from your radio will no longer control that servo (the one plugged into S7). 
You will have to move your servo to S8 to get that signal.

Here's another way to see the same problem.
Start with a 4 channel airplane and add 2 servos (Servo 5 and Servo 6). I mapped their outputs to RC channels 9 & 10
[image]
Plug in servos to S7 and S8 and verify that those radio channels correctly move those servos.
Now delete servo 5
[image]
You will be left with Servo 6, but it will move to output S7 in configurator
[image]
But moving that channel (in my case RC 10), it will move the servo on S8 and nothing will move the servo plugged into S7.

Also, if you change the servo number to something that would map it to an output that's larger than what you have on the controller, that signal won't come out any of the output ports. I changed mine to 10 and none of the servos on my FC will move, even though inav configurator shows it as port S7
[image]

If you look in the inav code at the function writeServos(), you will see that it uses the servo number to map to an output. It starts with the lowest servo number and maps that to the 1st output, then counts from there. (if your 1st servo number is 4, that gets mapped to the 1st servo output, but then it counts up from there. So if any servo numbers are skipped, those outputs are skipped)

MrD-RC, 2023-07-13:
In that case. Something had changed in the firmware. As you used to be able to skip index numbers. But it would not skip the output. It would act the same way as seen on the output mapping table. At some point there must have been a change in the firmware, which was not reflected in the configurator.

DzikuVx, 2023-07-14:
This is a known limitattion. Servos has to be defined sequencialy in order to firmware to correct assign outputs. And it's a firmawere nor configurator limitation. Thanks for reporting

closed 2023-07-14 by DzikuVx